prusa3d/PrusaSlicer #5230: [2.3.0-alpha4 - feature request]: Creating temp towers by hotend temp change in height modifier

Boby71, 2020-11-22:
### Version
2.3.0-alpha4 (and before)

### Operating system type + version
Win10/64 Bit

### 3D printer brand / version + firmware version (if known)
i3 MK2.5s - but it doesn't matter for this request

### Behavior
The idea is simple: If you would allow a change of print temperature in this dialogue...
[image]
...then it would be very, very easy to create a temp tower directly via the GUI without fiddling around in the gcode-section.
I don't know anything about what's under the hood, but it looks like it could be easily extended on this place.

_Is this a new feature request?_
YES

#### Project File (.3MF) where problem occurs
Not necessary

neophyl, 2020-11-22:
You can already do that using the add custom gcode facility in the same way that add colour change works.  Slice and go to the layer preview tab.  Select the layer in the preview where you want the temp change to happen, right click the little hexagon with the + symbol in it and select Add custom gcode.  Then add the gcode for the temp you want.  Can be done multiple layers to add the change temp commands for a temp tower.

jaapp, 2020-11-22:
I know this is not what you're asking, but It;s already extremely easy to generate a Temperature tower width one line of G-Code in the "Before layer change G-Code" section.

`M104 S{230 - (5 * int((layer_z / 8)))}`

This starts at 230 degrees and drops 5 degreed every 8 layers.

rtyr, 2020-11-24:
Already requested here https://github.com/prusa3d/PrusaSlicer/issues/2851, https://github.com/prusa3d/PrusaSlicer/issues/3635, https://github.com/prusa3d/PrusaSlicer/issues/3439.

As others mentioned, it is already very easy to create temperature tower.
[image]
[image]

Closing.

closed 2020-11-24 by rtyr

DiomedesDominguez, 2022-03-26:
I suggest to add both M104 and M109 commands if the 𐤃 temp is to big, like 6 or more degrees.