prusa3d/PrusaSlicer #5231: Fields in right-hand pane should not accept scroll input

mkoistinen, 2020-11-22:
### Version
2.3.0 alpha4+

### Operating system type + version
MacOS 11.0.1

### 3D printer brand / version + firmware version (if known)
Prusa MK3/S + MMU2/S

### Behavior
The right pane when in Platter mode contains Print, Filament, Printer settings as well as details of the models on the bed and other info. It is a long, scrolling pane. And the top of it is full of "combo boxes", which, have a nice down-caret to trigger. But, the fields themselves are scrollable too without requiring the down-caret button to be clicked first.

This creates the annoying condition where the user would like to scroll to the bottom to see, say, slicing details like duration of print, etc, uses the scroll function on their mouse, but instead triggers the Configuration Assistant or worse, simply changing their filament, print or printer settings without realizing it.

This behavior is exacerbated by the fact that now in the 2.3.0 alpha 4 beta, the top-most and other options in the filament menus (--- System Presets ---, --- User Presets ---), which are supposed to be a non-functional titles/separators, now open the Configuration Assistant when selected.

**The correct behavior is that these fields are only modifiable after clicking the down-caret button beside them and that the special menu separators do not do anything themselves.**

#### Project File (.3MF) where problem occurs
N/A

FidelCapo, 2020-11-26:
This change will be part of next release. Thank you for report.
Closing.

closed 2020-11-26 by FidelCapo

mkoistinen, 2020-11-29:
Seems to be fixed indeed in the 2.3.0 b1! Thank you!