prusa3d/PrusaSlicer #5232: Instant Segfault on starting prusa-slicer

mikelane, 2020-11-22:
In previous versions of the alpha, I had been able to build PrusaSlicer just fine. Starting about a week ago, every time I built the slicer and then start it using `src/prusa-slicer`. I decided to wait a bit to see if this was a temporary problem that would be fixed with subsequent commits. However, the problem still persists, so I'm trying to find a resolution (or to alert of a larger issue).

### Version
2.3.0-alpha4 

```bash
> git log
commit 2e0e63fed8f8a8c14f0938480f3f36bb3fa8f057 (HEAD -> master, origin/master, origin/HEAD)
Author: YuSanka <yusanka@gmail.com>
Date:   Sat Nov 21 18:15:20 2020 +0100

    Tooltips for the parameters labels on Settings tabs are back
```

### Operating system type + version
MacOS 11.1 Beta (20C5048k)

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S

### Behavior
Instant Segmentation Fault upon running the application after build. 

#### Steps Followed
1. From the PrusaSlicer directory, get latest changes using `git checkout master` and then `git pull`
1. Run `git clean -xdf`, just in case
1. Remove the build folders `rm -rf build` and `rm -rf deps/build`
1. Follow the guidance found in `doc/How to build - Mac OS.md`
    1. Run `brew upgrade` (`cmake`, `git`, and `gettext` have already been installed with brew and `brew upgrade` runs `brew update` automatically).
    1. `cd PrusaSlicer/deps`
    1. `mkdir build`
    1. `cd build`
    1. `cmake ..` (completes with no errors)
    1. `make` (completes with no errors)
    1. `cd ../..`
    1. `mkdir build`
    1. `cd build`
    1. `cmake .. -DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local"` (completes without errors)
    1. `make -j4` (completes without errors)
    1. `src/prusa-slicer`
1. There is log output after the `src/prusa-slicer` step and the GUI splash screen opens which has the 2.3.0-alpha4 version. 
1. The splash screen disappears and the following is found in the CLI log:

``` bash
❯ src/prusa-slicer
2020-11-22 08:55:33.547 prusa-slicer[4730:25502] CoreText note: Client requested name ".SFNS-Regular", it will get Times-Roman rather than the intended font. All system UI font access should be through proper APIs such as CTFontCreateUIFontForLanguage() or +[NSFont systemFontOfSize:].
2020-11-22 08:55:33.547 prusa-slicer[4730:25502] CoreText note: Set a breakpoint on CTFontLogSystemFontNameRequest to debug.
[2020-11-22 08:55:35.075480] [0x000000011e3c1e00] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 5): image not found
[1]    4730 segmentation fault  src/prusa-slicer
```

1. An OS crash reporter pops up and here are the system details:

```
Process:               PrusaSlicer [4730]
Path:                  /Users/USER/*/prusa-slicer
Identifier:            PrusaSlicer
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        zsh [1298]
Responsible:           iTerm2 [398]
User ID:               501

Date/Time:             2020-11-22 08:55:35.447 -0800
OS Version:            macOS 11.1 (20C5048k)
Report Version:        12
Bridge OS Version:     3.0 (14Y908)
Anonymous UUID:        6F900F34-3BBF-E84F-C4CF-027A400E89AA

Time Awake Since Boot: 310 seconds
[... 498 lines cut ...]
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       506.2M        7 
__OBJC_RO                         60.5M        1 
__OBJC_RW                         2448K        2 
__TEXT                           333.7M      332 
__UNICODE                          588K        1 
mapped file                      246.2M       28 
shared memory                       40K        4 
===========                     =======  ======= 
TOTAL                              1.5G     1309 

Model: MacBookPro13,3, BootROM 429.50.1.0.0, 4 processors, Quad-Core Intel Core i7, 2.6 GHz, 16 GB, SMC 2.38f11
Graphics: kHW_IntelHDGraphics530Item, Intel HD Graphics 530, spdisplays_builtin
Graphics: kHW_AMDRadeonPro450Item, AMD Radeon Pro 450, spdisplays_pcie_device, 2 GB
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x15A), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1675.1)
Bluetooth: Version 8.0.2f2, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.0 Bus
USB Device: Apple T1 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.2
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.2
```

bubnikv, 2020-11-23:
Does our official build crash for you?

we are compiling with the -DCMAKE_OSX_DEPLOYMENT_TARGET="10.9" parameter,
it may make a difference.

ne 22. 11. 2020 v 18:18 odesílatel Mike Lane <notifications@github.com>
napsal:

> In previous versions of the alpha, I had been able to build PrusaSlicer
> just fine. Starting about a week ago, every time I built the slicer and
> then start it using src/prusa-slicer. I decided to wait a bit to see if
> this was a temporary problem that would be fixed with subsequent commits.
> However, the problem still persists, so I'm trying to find a resolution (or
> to alert of a larger issue).
> Version
>
> 2.3.0-alpha4
>
> > git log
>
> commit 2e0e63fed8f8a8c14f0938480f3f36bb3fa8f057 (HEAD -> master, origin/master, origin/HEAD)
>
> Author: YuSanka <yusanka@gmail.com>
>
> Date:   Sat Nov 21 18:15:20 2020 +0100
>
>
>
>     Tooltips for the parameters labels on Settings tabs are back
>
> Operating system type + version
>
> MacOS 11.1 Beta (20C5048k)
> 3D printer brand / version + firmware version (if known)
>
> Prusa i3 MK3S
> Behavior
>
> Instant Segmentation Fault upon running the application after build.
> Steps Followed
>
>    1. From the PrusaSlicer directory, get latest changes using git
>    checkout master and then git pull
>    2. Run git clean -xdf, just in case
>    3. Remove the build folders rm -rf build and rm -rf deps/build
>    4. Follow the guidance found in doc/How to build - Mac OS.md
>       1. Run brew upgrade (cmake, git, and gettext have already been
>       installed with brew and brew upgrade runs brew update
>       automatically).
>       2. cd PrusaSlicer/deps
>       3. mkdir build
>       4. cd build
>       5. cmake .. (completes with no errors)
>       6. make (completes with no errors)
>       7. cd ../..
>       8. mkdir build
>       9. cd build
>       10. cmake ..
>       -DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local"
>       (completes without errors)
>       11. make -j4 (completes without errors)
>       12. src/prusa-slicer
>    5. There is log output after the src/prusa-slicer step and the GUI
>    splash screen opens which has the 2.3.0-alpha4 version.
>    6. The splash screen disappears and the following is found in the CLI
>    log:
>
> ❯ src/prusa-slicer
>
> 2020-11-22 08:55:33.547 prusa-slicer[4730:25502] CoreText note: Client requested name ".SFNS-Regular", it will get Times-Roman rather than the intended font. All system UI font access should be through proper APIs such as CTFontCreateUIFontForLanguage() or +[NSFont systemFontOfSize:].
[... 828 lines cut ...]
>     0x7fff30ab8000 -     0x7fff30b1afff  com.apple.CoreBluetooth (1.0 - 1) <B6A12956-3339-3093-9381-B5467B8B0992> /System/Library/Frameworks/CoreBluetooth.framework/Versions/A/CoreBluetooth
>
>     0x7fff30b1b000 -     0x7fff30b24fff  com.apple.SymptomDiagnosticReporter (1.0 - 79.40.12) <FEA66A4F-EAAF-3208-9042-25887D7191CE> /System/Library/PrivateFrameworks/SymptomDiagnosticReporter.framework/Versions/A/SymptomDiagnosticReporter
>
>     0x7fff30b39000 -     0x7fff30b45fff  com.apple.AppleIDAuthSupport (1.0 - 1) <99549167-35DB-3FB8-BF14-3B82379683B9> /System/Library/PrivateFrameworks/AppleIDAuthSupport.framework/Versions/A/AppleIDAuthSupport
>
>     0x7fff30b46000 -     0x7fff30beefff  com.apple.DiscRecording (9.0.3 - 9030.4.5) <6EE6F73A-7D76-345C-9C9D-82ED7D9F2935> /System/Library/Frameworks/DiscRecording.framework/Versions/A/DiscRecording
>
>     0x7fff30bef000 -     0x7fff30c22fff  com.apple.MediaKit (16 - 927.40.2) <CE80EC01-27D2-3A3B-8BCC-9A44CE416031> /System/Library/PrivateFrameworks/MediaKit.framework/Versions/A/MediaKit
>
>     0x7fff30c23000 -     0x7fff30d0efff  com.apple.DiskManagement (14.0 - 1733.60.1) <D212877D-FD3E-33A1-A758-9896AA45F2D9> /System/Library/PrivateFrameworks/DiskManagement.framework/Versions/A/DiskManagement
>
>     0x7fff310fc000 -     0x7fff3121cfff  com.apple.FileProvider (325 - 325) <3DEA1294-E441-3146-B99D-F934638B317A> /System/Library/Frameworks/FileProvider.framework/Versions/A/FileProvider
>
>     0x7fff3121d000 -     0x7fff31240fff  com.apple.GenerationalStorage (2.0 - 322) <3CBBFDE8-5BD8-3FD9-AD59-46735AEA1623> /System/Library/PrivateFrameworks/GenerationalStorage.framework/Versions/A/GenerationalStorage
>
>     0x7fff31b9f000 -     0x7fff31bdbfff  com.apple.DebugSymbols (195.1 - 195.1) <06D6316F-960C-349D-ADD4-EECAE9EC7514> /System/Library/PrivateFrameworks/DebugSymbols.framework/Versions/A/DebugSymbols
>
>     0x7fff31bdc000 -     0x7fff31c99fff  com.apple.CoreSymbolication (12.2 - 64541.4) <43A58BA8-8D05-353A-96FC-55F5A4CB3105> /System/Library/PrivateFrameworks/CoreSymbolication.framework/Versions/A/CoreSymbolication
>
>     0x7fff32b2f000 -     0x7fff32b92fff  com.apple.framework.Apple80211 (17.0 - 1728) <16528373-ACFD-3FCD-AC06-60FF9CD16C05> /System/Library/PrivateFrameworks/Apple80211.framework/Versions/A/Apple80211
>
>     0x7fff32b93000 -     0x7fff32ce5fff  com.apple.CoreWiFi (3.0 - 341) <A68A3EB2-2937-34A7-BA30-4F7B43CCC2E3> /System/Library/PrivateFrameworks/CoreWiFi.framework/Versions/A/CoreWiFi
>
>     0x7fff32f52000 -     0x7fff32f61fff  com.apple

Spybyte, 2020-11-23:
Same happened to me with the official alpha4 build. I just designed a model which crashes the moment I click on slice. Other models work without a flaw. I can provide the stack trace as well the model I try to slice (it works with PrusaSlicer-2.2.0+) if needed.

mikelane, 2020-11-23:
> Does our official build crash for you? we are compiling with the -DCMAKE_OSX_DEPLOYMENT_TARGET="10.9" parameter, it may make a difference. ne 22. 11. 2020 v 18:18 odesílatel

The distributed alpha 4 version does not crash, so I'll try it with that target and let you know. Thanks.

mikelane, 2020-11-26:
> Does our official build crash for you? we are compiling with the -DCMAKE_OSX_DEPLOYMENT_TARGET="10.9" parameter, it may make a difference. ne 22. 11. 2020 v 18:18 odesílatel 

Using a deploment target of 10.9 didn't build. There was a warning that some option was not valid until 10.12. I set the deployment target to 10.12 and it built fine, but it segfaulted immediately upon running.

bubnikv, 2020-11-26:
Thanks for heads up. We are still waiting for our Apple Silicon with
deployment target 11. Once we get our hands on it, we will likely reproduce
your issue.

mikelane, 2020-11-27:
> Thanks for heads up. We are still waiting for our Apple Silicon with deployment target 11. Once we get our hands on it, we will likely reproduce your issue. čt 26. 11. 2020 v 1:01 odesílatel 

FYI, I have an intel machine:

```
MacBook Pro (15-inch, 2016)
Processor: 2.6 GHz Quad-Core Intel Core i7
```

... running Big Sur:

```
macOS Big Sur
Version 11.1 Beta (20C5048k)
```

obelisk, 2020-11-29:
I triaged this today, the cause is outdated wxWidgets

```
➜  src git:(d5a6602f1) ✗ lldb ./PrusaSlicer
(lldb) target create "./PrusaSlicer"
Current executable set to '/Users/obelisk/Documents/repos/cpp/PrusaSlicer/build/src/PrusaSlicer' (arm64).
(lldb) r
Process 96677 launched: '/Users/obelisk/Documents/repos/cpp/PrusaSlicer/build/src/PrusaSlicer' (arm64)
[2020-11-28 17:13:38.872135] [0x0000000102953d40] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 5): image not found
Process 96677 stopped
* thread #1, name = 'slic3r_main', queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x8)
    frame #0: 0x0000000101163e54 PrusaSlicer`wxMacCoreGraphicsPenBrushDataBase::CalculateShadingValues(info=0x0000000111e17748, in=0x000000016fdf70b0, out=0x000000016fdf7110) at graphics.cpp:627:33
   624 	    if (f <= 0.0)
   625 	    {
   626 	        // Start
-> 627 	        out[0] = stops.comps[0].red;
   628 	        out[1] = stops.comps[0].green;
   629 	        out[2] = stops.comps[0].blue;
   630 	        out[3] = stops.comps[0].alpha;
Target 0: (PrusaSlicer) stopped.
```

If you want it to build correctly change the deps from pulling from http://GitHub.com/prusa3d/wxWidgets to https://github.com/wxWidgets/wxWidgets in ./deps/wxWidgets/wxWidgets.cmake

bubnikv, 2020-12-07:
Our dependencies script for Big Sur / M1 pulls patched wxWidgets 3.1.4. I expect this to fix the crash on X86 Big Sur as well if you change your dependency script to pull the same wxWidgets versiojn.

We are building for OSX Maveric and newer, thus we build against an older SDK and that works.
Closing.

closed 2020-12-07 by bubnikv

specialistvlad, 2020-12-31:
So is there any way to build it? I would like to decrease `SCALING_FACTOR` because I trying to found solution for bigger printing size and can't make any experiment right now. The situation is very similar to topic starter. I can build it like it readme but I see segmentation. If I use recommendation `cmake .. -DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local" -DCMAKE_OSX_DEPLOYMENT_TARGET="10.9"` it gives me an error 

```/Users/vk/prj/test/PrusaSlicer/deps/build/destdir/usr/local/include/boost/core/uncaught_exceptions.hpp:105:45: error:
      'uncaught_exceptions' is unavailable: introduced in macOS 10.12
    return static_cast< unsigned int >(std::uncaught_exceptions());
                                            ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/exception:130:63: note:
      'uncaught_exceptions' has been explicitly marked unavailable here
_LIBCPP_FUNC_VIS _LIBCPP_AVAILABILITY_UNCAUGHT_EXCEPTIONS int uncaught_exceptions() _NOEXCEPT;`

deckar01, 2021-04-22:
Switching to the wxWidgets repo without the patch caused my build to fail. Forcing `v3.1.4-patched` fixed the issue. The old comment stated that 3.1.4 was required for Big Sur, but the check only tested that the architecture was Apple ARM, which failed to handle Big Sur on other architectures.

This has been fixed with 264ce71e and d7a786e8, but hasn't made it into a stable tag yet.