prusa3d/PrusaSlicer #5233: 3DConnexion 3D SpaceMouse Jumpy Pan moves on Linux

Area5142, 2020-11-22:
### Version
PS 2.3.0 Alpha4 and earlier versions

### Operating system type + version
Linux Mint 19.3 x64

### 3D printer brand / version + firmware version (if known)
Not needed

### Behavior
When using the 3DConnexion 3D Space Mouse the rotation movements is smooth, but pan movements is jumpy or jerky.

This has been true from PS Version 2.2.0, but I only recently got a screen capture program installed to capture the problem.

YouTube video of the problem: [PrusaSlicer 3DConnexion Mouse Jerky Movements on Linux](https://www.youtube.com/watch?v=etIJj260UBM)

In the video I try to only move one axis at the time. When moving more than one axis the rotation is still smooth and all pan movements is jerky/jumpy.

Sometimes the pan becomes smooth for a short period and then returns to jumping.

_Expected behavior:_
Pan movements should be smooth like in rotation.

When using the same 3DConnexion Mouse in other programs on the same PC, all motions is smooth (FreeCAD, Blender, ...).

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Not Needed - just use empty build plate or make a cube as I did.

bubnikv, 2020-12-15:
Fixed with aa8f97fce261bcc433a989d6a7f9d38080f6cccc

closed 2020-12-15 by bubnikv

lukasmatena, 2020-12-18:
This should be fixed in [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1). Let us know if it isn't. Thanks.

maximlevitsky, 2021-01-01:
This is indeed fixed! (I just bought the 3d mouse and I already enjoy it)

Area5142, 2021-01-02:
Yes. this is fixed in 2.3.0 RC2.