prusa3d/PrusaSlicer #5234: MMU2s - Converting the purge block from/to Imperial units crashes the application

Area5142, 2020-11-22:
### Version
PS 2.3.0 Alpha4

### Operating system type + version
Linux Mint 19.3 x64

### 3D printer brand / version + firmware version (if known)
Prusa MMU2S

### Behavior
Right click the purge block:
[image]

An option to _Convert from Imperial units_ and _Revert conversion from Imperial units_ is shown. Selecting this option crashes the program.

_Expected behavior:_
Why should the purge block need unit conversion at all. The size is partly calculated or set elsewhere in the units selected for the program. 

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Not Needed - just use empty build plate and start MMU2s project.

bubnikv, 2020-11-23:
@enricoturri1966 would you please disable the pop-up menu on the wipe tower? Thanks.

bubnikv, 2020-11-24:
Will be fixed in the next release. Closing.

closed 2020-11-24 by bubnikv