prusa3d/PrusaSlicer #5235: QOL suggestion - dark mode option for settings menus

3dGeneDesigns, 2020-11-23:
### Version
_2.3.0-alpha3 and under_

### Operating system type + version
_Windows 10/64bit_

### 3D printer brand / version + firmware version (if known)
_i3 Mk3 / MMU2 / custom_

### Request
_A dark mode for the three setting tab backgrounds. Going from the 'plater' tab to one of the three other tabs can be quite blinding if you're like me and work during the night when it's naturally darker. I think it'd be nice to maybe have a consistent grey across all tabs and have it be a toggle in preferences. Maybe the side toolbar as well since that too has a white background._

[image]

Before blindness:
[image]

Blindness if in dark environment:
[image]

bubnikv, 2020-11-23:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4493
Please bug Microsoft to provide dark mode for Win32 controls.

closed 2020-11-23 by bubnikv

Kollimolli, 2020-11-25:
Maybe Microsoft never fixes dark mode APIs. Perhaps it is better develop an alternative way to make dark mode possible? Currently i have installed many apps that have dark mode/theme and all them works well, so there must be an another way to make dark mode/theme.

"Don’t sit and wait forever for the mountains to move out of the way. Use a detour"