prusa3d/PrusaSlicer #5236: Color not correctly shown after slice

morimorida, 2020-11-23:
### Version
Prusa Slicer 2.3.0 Alpha 4
### Operating system type + version
Windows 10 64bit

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S + MMU2S
### Behavior
When object is sliced, it shows a different color from what it should be based on the filament color used.

#### Project File (.3MF) where problem occurs
Picture attached.
[image]
[image]

neophyl, 2020-11-23:
What colours do you have assigned to your 4 extruders ?  I notice in the preview you have 'Tool' selected as view type so isn't it using the colours you have assigned to the various tools ?

FidelCapo, 2020-11-23:
@morimorida Can you please attach 3mf file? Thank you

morimorida, 2020-11-23:
@neophyl , as you can see from PrusaSlicer 2.2.0 version, it shows correctly even if it's on "Tool" as a view type.
[image]

@FidelCapo , I have attached the 3mf file for your review.
[Kobayashi_Fidget_Cube-v1.2.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5584206/Kobayashi_Fidget_Cube-v1.2.3mf.zip)

bubnikv, 2020-11-24:
It will be fixed in PrusaSlicer 2.3.0-beta. Closing.

closed 2020-11-24 by bubnikv