prusa3d/PrusaSlicer #5237: Strange artifacts on flat surfaces

alexiri, 2020-11-23:
### Version
2.2.0+linux-x64

### Operating system type + version
Ubuntu 20.04

### 3D printer brand / version + firmware version (if known)
Prusa MK3 stock

### Behavior
Prusa Slicer introduces strange artifacts on a flat part of the model.

[image]
[image]

Layer 2.20 looks ok:

[image]

Layer 2.40 starts to look kind of weird:

[image]

Layer 2.60 is just wrong:

[image]

#### Project File (.3MF) where problem occurs
[v6-Fixed-Power-Supply-v8-Front-Front.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/5583318/v6-Fixed-Power-Supply-v8-Front-Front.3mf.gz)

neophyl, 2020-11-23:
I suspect its becayse its not 100% perfectly flat and the vertices lie right on the edge.  Most of them are 2.49999 relative z and others are 2.50000 z according to blender.  

Does the following geometry pattern look familiar ?  It should as that's where the extra layers match.
[image]

If you slice it at 0.15 layer height it slices flat.  That is down to the slicers decision for all those top vertices falls within the same layer.  With them being 2.4999-2.5 its putting some areas with an extra layer, hence your slicing 'problem'.

Either add a modifier at that level to set the layer height to 0.15 or fix the model.

bubnikv, 2020-11-23:
https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)#strange-pattern-on-a-flat-surface

alexiri, 2020-11-23:
I see, thank you both and sorry for the noise!

closed 2020-11-23 by alexiri

bubnikv, 2020-11-23:
no problem