prusa3d/PrusaSlicer #5240: Slicing s-buffer-spools.stl (MMU2 part) crashes PrusaSlicer

plampix, 2020-11-23:
### Version
2.3.0-alpha4+linux-x64 (AppImage from the releases page

### Operating system type + version
Debian GNU/Linux 10

### 3D printer brand / version + firmware version (if known)
Not relevant

### Behavior
When I try to open the file downloaded from your github repository at:
https://raw.githubusercontent.com/prusa3d/Original-Prusa-i3/MMU2/STL/s-buffer-spools.stl

PrusaSlicer crashes after slicing the model.

The PrusaSlicer 2.2.0 AppImage correctly slices the stl, and does not crash.

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs

[s-buffer-spools.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/5584561/s-buffer-spools.3mf.gz)

plampix, 2020-11-24:
This does not appear to be the issue that was fixed in 62bdc192d87a9a1835e6ff6ff10b5a0893279c98

bubnikv, 2020-11-24:
Fixed with 62bdc192d87a9a1835e6ff6ff10b5a0893279c98

closed 2020-11-24 by bubnikv

plampix, 2020-11-24:
I'm sorry, had to do a clean build to see the issue has indeed been solved.

Great work!