prusa3d/PrusaSlicer #5241: Vertical slider time estimate doesn't match total print time

ratsey, 2020-11-23:
### Version
2.3.0-alpha4

### Operating system type + version
WIN 10

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro / Marlin

### Behavior
On slicing, print time shows correctly, but is not reflected in the layer time estimates. I would expect the top layer value to match the total print time.

_Vertical slider in the G-code Preview (FDM mode) and in the layer view (SLA mode) newly shows a ruler with Z-heights. A time estimate for the respective layer is shown when hovering the mouse over the slider area._

[image]

Kachidoki2807, 2020-11-24:
Hi, the text in grey at the right of the slider is millimeters.

> A time estimate for the respective layer is shown **when hovering** the mouse over the slider area.

You have to place your mouse onto the ruler to see the ETA.

Regards.

ratsey, 2020-11-24:
Top-most value doesn't match the item height, but it's close, so I'll take that. I don't get a time on mouse-over though, it just says 'time'. I assume then that's the default value for a variable that's not being updated:

[image]

Kachidoki2807, 2020-11-24:
I can't reproduce this behaviour. Would you be so kind to share the .3mf?

ratsey, 2020-11-24:
@Kachidoki2807 .3mf attached

Kachidoki2807, 2020-11-24:
Thx @ratsey 

I reproduced the bug with your 3mf. So there are two points:

> Top-most value doesn't match the item height, but it's close

This value is rounded to the nearest "multiple" of layer height, including the first layer.

> I don't get a time on mouse-over though, it just says 'time'. I assume then that's the default value for a variable that's not being updated

It seems to be "support" related. Disabling the support bring back the time estimation. This is a bug.

Kachidoki2807, 2020-12-22:
Hello,
I tried again with the 2.3.0-rc1, it seems to be solved.
@ratsey would you be so kind to confirm that this bug is fixed on your side?

rtyr, 2021-01-28:
No response. It should be fixed. Closing.

closed 2021-01-28 by rtyr