prusa3d/PrusaSlicer #5242: Glitched GUI controls on Linux in alpha4

VorpalBlade, 2020-11-23:
### Version
PrusaSlicer 2.3.0 alpha4+linux-x64 (Linux AppImage)

### Operating system type + version
Ubuntu 18.04, using the Cinnamon desktop environment.

System Info: [prusa-sysinfo.txt](https://github.com/prusa3d/PrusaSlicer/files/5585258/prusa-sysinfo.txt)

### 3D printer brand / version + firmware version (if known)
Not relevant I think, but Prusa MK3s.

### Behavior
Notice weird glitch above the top of the layer slider in this screenshots:
[image]

To reproduce:
* In GCode preview, move the mouse over layer selecting slider.
* A dotted outline appears, but at the top it instead looks like garbage.

Additional notes of interest:
* This also happens in the standalone gcode viewer.
* This does **not** happen for the slider at the bottom (the dotted outline looks fine on all borders).
* The dotted outline looks fine in alpha3 (GTK2) and in version 2.2.0.

_Is this a new feature request?_
No, this is a bug.

#### Project File (.3MF) where problem occurs
Happens for every file i tried, including a 3D Benchy.

rtyr, 2021-02-25:
Does it happen with final 2.3.0?

VorpalBlade, 2021-02-27:
I just checked. It appears to be fixed! However I noticed another issue, which is that the layer heights get cut off for tall objects. I'll open a separate bug for that.

closed 2021-02-27 by VorpalBlade