prusa3d/PrusaSlicer #5245: Crashes when slicing with standard 0.10 detail settings

19Ted54, 2020-11-24:
### Version
2.3.0-alpha 4

### Operating system type + version
Windows 10 Home
2004
08.11.2020
19041.630

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S
3.9.1
NO MODS

### Behavior
Crashing when slicing with standard 0.10 detail settings. Frequent crashes with other settings and other files...
When crashed, start again, use same settings and file > crashing

#### Project File (.3MF) where problem occurs

[Black_Parts.zip](https://github.com/prusa3d/PrusaSlicer/files/5588309/Black_Parts.zip)

pilotltd, 2020-11-24:
Yep - crashes for me on slicing with 0.15 detail.

bubnikv, 2020-11-24:
Fixed with 62bdc192d87a9a1835e6ff6ff10b5a0893279c98

closed 2020-11-24 by bubnikv