prusa3d/PrusaSlicer #5246: [2.3.0-alpha4] Crash after slicing cut object

zeroflow, 2020-11-24:
### Version
2.3.0-alpha4

### Operating system type + version
Windows 10 20H2

### 3D printer brand / version + firmware version (if known)
Mk3S + MMU2S

### Behavior
1) Import STL
2) Cut STL (options don't matter)
3) Slice Model -> Crash

This issue does _not_ happen with the default settings (0.2mm Speed, Mk3S, ...) but only with the settings in the uploaded 3MF file.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[Filament Spacer v2.zip](https://github.com/prusa3d/PrusaSlicer/files/5588630/Filament.Spacer.v2.zip)

zeroflow, 2020-11-24:
I have bisected this problem in the exported 3mf files.

The broken settings with this cut model is
; fill_density = 10%
; fill_pattern = gyroid

Others like rectilinear or grid @ 10% infill works.
Gyroid @ 20%, 15% and 5% works, only 10% fails.

neophyl, 2020-11-24:
As I found with issue #5250 if you use those setting but set the new anchored value to 0 then it will slice.  As soon as the anchored value is set to even 1mm or a % then it seems to crash.

bubnikv, 2020-11-24:
Fixed with 62bdc192d87a9a1835e6ff6ff10b5a0893279c98

closed 2020-11-24 by bubnikv