prusa3d/PrusaSlicer #5247: mm3_per_mm() negative value when generating supports on PrusaSlicer 2.2.0

MG4mes, 2020-11-24:
### Version
Version 2.2.0+win64

### Operating system type + version
Windows 10 (build 19041), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Ender 3 with SKR mini E3 V2 using Marlin bugfix-2.0.x from 2020-07-15

### Behavior
* _Describe the problem_
I am not able to generate supports everywhere or on build plate only (manually nor automatically) and only works if I choose 
"For support enforcers only" or increase the Supports extrusion width to a high value (0.6mm)

* _Steps needed to reproduce the problem_
Not sure but setting the Supports extrusion width equal to the default extrusion width (0.4) or a value close to it causes this issue on my end

* _Expected Results_
Generating supports successfully without any errors

* _Actual Results_
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[Ender-3_filament_guide_v2.2.zip](https://github.com/prusa3d/PrusaSlicer/files/5589016/Ender-3_filament_guide_v2.2.zip)

I am new to Github so please inform me if anything is missing.
Thanks.

bubnikv, 2020-11-24:
You configured an unreasonable maximum layer height. 0.25mm is likely more what you want. The support generator tries to extrude with the maximum allowed layer height and it fails.

[image]

MG4mes, 2020-11-24:
Oh for some reason I thought it was 0.28mm *sigh*
Thank you!

closed 2020-11-24 by MG4mes