prusa3d/PrusaSlicer #5248: Smart temperature management on multiple extruder machines

Kachidoki2807, 2020-11-24:
Hello,
This is a **new feature request proposal**.

PrusaSlicer 2.3.0-alpha4 introduce a great new Z-Ruler with the ETA up the the hovered layer. That's gave me an idea I want to share with you.

On a multiple extruder system, like the more and more popular IDEX architecture, it would be great to be able to heat up each nozzle only at the right time.

There is already an existing "temperature variation" parameter, but it is not a smart enough behavior.
As PrusaSlicer is now able to compute precisely the time between layers (for the ruler), there is a small step left to add the ability **to know the time between two toolchanges**.

I mean, if we replace the "temperature variation" by a new temporal parameter: "reheating time" in seconds, PrusaSlicer could be able to completely shutdown the heating of the unused extruder, and when coming at x seconds from the next toolchange, reheat it.

This "reheating time" parameter would be static and set like a maximum time, that will probably often reheat too soon. But it is easy to implement (I guess) and  more _**smart and efficient**_ than the current system. We probably don't want to go to a complex °C/s cooling and heating curves computation.

This is roughly the idea, what do you think?

Kachidoki2807, 2020-12-14:
Hello,

I was fighting with PrusaSlicer since some times to make it working smoothly with an IDEX printer. I am able to get something usable (excepted the too big purging volume I have to reduce manually each time and the stability of tall wipe towers), and my only leaving blocking issue is the long time parking that make some filament messy. PolySupport really don't like to be kept hot during one, two or a lot of hours waiting to be extruded.

Cause I cannot tell to the future users of this machine they have to tinker the settings and as I am accumulating issues (there are several open issues, like bad fan management), I decided to look elsewhere, especially IdeaMaker which is dedicated to IDEX machines. I am still discovering this slicer, I already found a bug (M605 Sx make the preview crazy), but globally it already take into account some details for which I had to find a workaround into PS.

Regarding this specific feature request, I found out that this feature already existed into IdeaMaker.

If the Prusa XL will include a toolchanger that is not a MMU, I am sure you will encounter and beat these issues.

Regards.

n8bot, 2020-12-22:
I would like this feature to. Copy/pasted from my duplicate issue:

PrusaSlicer has very accurate print time calculation built in. Using this ability, it should be fairly simple to implement a feature to cool down and heat up tools so that they are ready for use when they are needed. The feature is intended to reduce the time taken waiting for tools to heat up at tool changes.

The slicer IceSL has this feature. They describe it like this:

> Active temperature control is specifically designed to avoid material to degrade and ooze from idle extruders. The idle extruder is dynamically cooled down and heated during printing, such that it is ready when needed and cooled when not needed. Adjust extruder parameters idle_extruder_temp_degree_c and extruder_degrees_per_sec when using this feature.

IceSL implements this feature well. A target standby temperature is set very low: 100 degrees C by default. The user then estimates and inputs the time it takes to heat/cool in degrees per second. The slicer then calculates, for all tool changes, at what point during the layer to start pre-heating the next tool. Also, importantly, it decides to what temperature to cool the tool being deselected so that it doesn't cool down too much for the time it needs to be inactive. This means the tools may be commanded to different temperatures for standby depending on how long it will be inactive. It gets as cool as possible, given the time, and with enough time will be cooled down much lower than other slicers achieve.

This is only really useful for printers with discrete tools: IDEX or toolchangers. Prusa may not have any such printers on the market right now, but they should make an IDEX printer or a toolchanger because they are friggin rad.

Also, if Prusa Research is not willing to implement this feature, I would be glad to help in any way I can, and seek help form others to implement this feature in a way that would be acceptable to the Prusa team.

jaddel, 2022-10-21:
Is there currently any kind of workaroud to implement such thing for an IDEX printer?

awabom, 2023-07-06:
@jaddel 
Yes, I've made some python scripts that tries to do a smart pre-heat of the nozzle.

https://github.com/awabom/3dprint/

It's called 'IDEXPreheat', mostly done for my SV04 but should work with other printers.

Mike-Flexicon, 2025-01-16:
> @jaddel Yes, I've made some python scripts that tries to do a smart pre-heat of the nozzle.
> 
> https://github.com/awabom/3dprint/
> 
> It's called 'IDEXPreheat', mostly done for my SV04 but should work with other printers.

You inspired me to create a python script that would also work for reprap firmware printers.

The problem with reprap is, you cannot leave the timestamps in the gcode as they are not supported. So the code a friend helped me write adds the preheat and then removes all of the time stamps.  OrcaSlicer now supports this preheat, but it is done incorrectly for reprap printers as well. They set the active temperature of the tool they intend to warm up, not the standby temp. My script also sets the soon to be inactive tools new lower standby temp.

I see I cannot attach a python script here so not sure how to share it. I'm fairly new to github.

closed 2026-09-29 by prusaslicer-bot[bot]