prusa3d/PrusaSlicer #5249: Text on Slice button

towlerg, 2020-11-24:
### Version
All versions to data

### Operating system type + version
Win 7 x64

### 3D printer brand / version + firmware version (if known)
unrelated - internal to PrusaSlicer

### Behavior
After "Slice Now" is pressed, text immediately changes to "Export GCode" even though slicing has not completed. Perhaps text could read "Slicing" until slicing complete or alternately disable the button ie grey out the text.

#### Project File (.3MF) where problem occurs
unrelated - internal to PrusaSlicer

neophyl, 2020-11-24:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3939

bubnikv, 2020-11-24:
> Perhaps text could read "Slicing" until slicing complete or alternately disable the button ie grey out the text.

This is the way it works in simple mode. In advanced and expert mode you may plan the G-code export before the slicing finishes.

closed 2020-11-24 by bubnikv

towlerg, 2020-11-25:
Thanks for getting back to me. I see your argument but I find it hard to believe that anyone would want to skim a few seconds initiating a process who's duration is measured in hours. I see that users unused to PrusaSlicer (way to go by the way, getting better o every iteration) should not be in expert mode but if you find that unfamiliar software has an expert mode, you select it. Not so much human nature as geek nature.