prusa3d/PrusaSlicer #5250: [2.3.0-alpha4] Crash after slicing object with less than 15% infill

haplm, 2020-11-24:
### Version
2.3.0-alpha4+

### Operating system type + version
macOs Big Sur, 11.0.1, Intel

### 3D printer brand / version + firmware version (if known)
Original MK3S, stock, latest firmware

### Behavior
* When I try to slice attached object with 15% and less infill, slicer crashes. The issue is most likely related to #5246 

* _Steps needed to reproduce the problem_
  * Import object
  * Set material to Fillamentum PLA (perhaps unrelated)
  * Set infill to 15% or less, make sure that gyroid is used, layer height 0.2 Quality
  * Hit Slice. Slicer will crash
* _Expected Results_
  * Slicer doesn't crash
* _Actual Results_
  * [crash_ps.txt](https://github.com/prusa3d/PrusaSlicer/files/5589723/crash_ps.txt)

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[SphereFlatSides.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/5589724/SphereFlatSides.stl.zip)

neophyl, 2020-11-24:
It seems to be related to the anchored infill option.  If set to 0 (not anchored it slices fine, as soon as you set it to even 1mm and slice it crashes.

yodine, 2020-11-24:
I have the same crash on MAC with some STL with default settings: gyroid and 15% infill. As seen in related crash, going other 15%, removing anchored or change infill type make it slide.
Here is a very simple hole cylinder shape: the taller crashes and the smaller won't. Exact same base, just the height changes.
[HoleCylinder.zip](https://github.com/prusa3d/PrusaSlicer/files/5590304/HoleCylinder.zip)

bubnikv, 2020-11-24:
Fixed with 62bdc192d87a9a1835e6ff6ff10b5a0893279c98
Closing.

closed 2020-11-24 by bubnikv