prusa3d/PrusaSlicer #5251: PrusaSlicer 2.3.0 (alpha4) remove some parts when slicing

AndWik, 2020-11-24:
### Version
_Version of PrusaSlicer used goes here_
PrusaSlicer 2.3.0 (alpha4)

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
PrusaSlicer-2.3.0-alpha4+win64-202011211615.zip

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 10 Pro 64

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
MK3S FW 3.9.1

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

[image]
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

AndWik, 2020-11-25:
Hi

For some reason I was unable to upload the 3MF file.
I have put the scad, stl and 3fm-files on my personal WEB-server

I has happened twice only. Last time i could arrange the object differently. This time I could not.
In a second version I changed the bottom of the object. Then it came out correct.
The stl seems ok. Cura and Simplify3D sliced the object correctly.
Hope this is helpful to find the issue

Thanks
Anders
Sweden

neophyl, 2020-11-25:
I believe you have to zip up any of those file types to attach them to GitHub.  

Ps also cant see a link to the files on your server in the post above

foreachthing, 2020-11-25:
SCAD? Is your model manifold? Please check your model for errors (PS will let you know if there are any).

AndWik, 2020-11-25:
http://fuzzyfool.no-ip.com/Prusa3D/

Sorry

foreachthing, 2020-11-25:
Your model has issues:
[image]
Left: Fixed with Blender. Right: your STL.

If you don't have Blender or don't know how to use it, you can fix your model through https://service.netfabb.com/.
If you download a release of PS, you get netfabb with it - not if you compile it yourself.

AndWik, 2020-11-25:
OK Thanks

closed 2020-11-25 by AndWik