prusa3d/PrusaSlicer #5252: Add 'thumbnail' option into physical printer configuration

haplm, 2020-11-24:
Before #4112 is implemented, it would be great if configuration option `thumbnails =` could be specified in physical printer configuration file.

It seems really redundant to clone built-in configuration for MK3 just to be able to enable thumbnails, and then re-clone that every time there's a configuration update...

_Is this a new feature request?_
Yes

bubnikv, 2020-11-24:
I don't understand your use case.

haplm, 2020-11-25:
It is relatively straightforward ;-)

I'm using Octoprint + OctoDash with 'PrusaSlicer Thumbnails' plugin. That allows me to display thumbnails even for the jobs that are printed on MK3S on my OctoDash screen. I just need to be able to insert thumbnails into gcode sliced for MK3S, and I don't want to clone the whole printer profile because of that. To me, the physical printer config file looks like a natural place for such a configuration item.

bubnikv, 2020-11-25:
thanks, now I understand. By the way I believe the MK3S firmware has an
issue with the thumbnails when reading from an SD card.

haplm, 2020-11-25:
Thanks, I've read about that, however everything seems to be just fine when printing from OctoPrint.

morphias2004, 2020-12-03:
> Before #4112 is implemented, it would be great if configuration option `thumbnails =` could be specified in physical printer configuration file.
>
> It seems really redundant to clone built-in [...]

100%. Please add this!  👍

closed 2026-09-29 by prusaslicer-bot[bot]