prusa3d/PrusaSlicer #5254: Unexpected Behavior @ Application Open and When GCODE is Generated

TJKB2TQU, 2020-11-24:
First off, I love some of the new features! OK, down to it..... Data:
PrusaSlicer Version:
Version:   2.3.0-alpha4+win64
Build:     PrusaSlicer-2.3.0-alpha4+win64-202011211615

My device is running:
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 17,011MB
OpenGL installation
GL version:   4.6.0 - Build 27.20.100.8681
Vendor:       Intel
Renderer:     Intel(R) HD Graphics 520
GLSL version: 4.60 - Build 27.20.100.8681

This issue is not with the Print. The prints are fine. It is unexpected behavior when I open PrusaSlicer, then another when a GCODE is generated. 

1) I get a GUI Pop-up that indicates it cannot find an "Icon" in the apps .exe file. 
See image:
[image]

2) Second I can close out that GUI Pop-up and the PrusaSlicer app works fine. It does however generate another pop-up DOS window that indicates it had an error which reads:
[2020-11-24 11:49:33.831685] [0x0000280c] [error]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
NOTE: This Error is generated every time a .GCODE file is generated. If the Pop-up DOS Window is closed, the entire application closes. If I continue to work and generate GCODE's, there are no other issues, other than it logs the error for every GCODE generated. Again, prints are fine. 
See image:
[image]

If you have any questions or require additional information, please feel free to contact me.

Thank you in advance for your attention.

Regards,

Anthony
alamirat@optonline.net

bubnikv, 2020-11-28:
The 2nd issue will be fixed in PrusaSlicer 2.3.0-beta1.

We don't know how to reproduce the 1st issue. Would you please retest with PrusaSlicer 2.3.0-beta1? Thanks.

TJKB2TQU, 2020-11-28:
Hi, 

Thanks for the reply! The 1st issue, if you read the GUI message, it sates it cannot fond the Icon in the .exe file. It shows the path that it is looking for the .exe file in, but the search location is NOT where the files are stored.  Hope that helps...

> 
> 
>      
> 
>     The 2nd issue will be fixed in PrusaSlicer 2.3.0-beta1.
> 
>     We don't know how to reproduce the 1st issue. Would you please retest with PrusaSlicer 2.3.0-beta1? Thanks.

Anthony Lamirata
516.232.3887

bubnikv, 2020-12-03:
The 1st issue shall be fixed with 6fd1143b1caa2d55bb3468a79c3373a5c54fdd2f
in PrusaSlicer 2.3.0-beta2.
Closing.

closed 2020-12-03 by bubnikv