prusa3d/PrusaSlicer #5256: Can not Export or Slice

dedike, 2020-11-24:
Version: 2.2.0+win64
Build: PrusaSlicer-2.2.0+win64-202003211145
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 8,485MB

After pressing Export G-code I get an error message which says:

G-code export to C:\Users\Gergo\AppData\Local\Temp\.7052.gcode failed. Cannot open the file for writing.

FRESH INSTALL, STOCK WINDOWS WTF

fe60, 2020-11-25:
>  Cannot open the file for writing.

Sounds like an issue with the file access on the level of the operating system.
Does the export to another folder work (e.g. C:\Users\Gergo\Desktop)?

neophyl, 2020-11-25:
I'd try running PS as admin and see if that makes any difference.

dedike, 2020-12-01:
Hi, no it doesn't :/

bubnikv, 2020-12-02:
Could you copy files to C:\Users\Gergo\AppData\Local\ with windows explorer?
Could you save text files with notepad into the same folder?
Are you having multiple PrusaSlicer instances running?
Do you have some 3rd party antivirus installed? Or any other 3rd party applications that could block the access?

t0mx, 2021-03-22:
Is here anything new? I have the same issue and cannot find any further information. A workaround would be fine.

bubnikv, 2021-03-22:
@t0mx thanks for heads up, however your issue may be different from the others. How did your issue manifest?

lukasmatena, 2022-04-05:
No response from either @dedike or @t0mx. No similar issues reported in a year, hopefully not a problem in PrusaSlicer. Closing.

closed 2022-04-05 by lukasmatena

dedike, 2022-04-05:
I had some very weird default folder restrictions on the prusa folders in appdata. Once I changed those all worked fine. :)

bubnikv, 2022-04-06:
@[dedike](https://github.com/dedike)

> I had some very weird default folder restrictions on the prusa folders in appdata. Once I changed those all worked fine. :)

Would you please give us more detail? How did that happen? How come that only PrusaSlicer was affected?