prusa3d/PrusaSlicer #5257: Program Crash when Slicing at .10

Terry-Pro, 2020-11-24:
### Version
Version 2.3.0 Alpha 4

### Operating system type + version
Windows 10 Pro 64-bit build 19041.630

### 3D printer brand / version + firmware version (if known)
Prusa Mk 3S stock. Firmware 3.9.1-3518

### Behavior
* When slicing any object at ".10 Detail" layer height, Alpha 4 attempts the slice but crashes/shuts down after approximately 5 seconds. 
* Happens with any object .stl in Alpha 4 when I select .10 layer heights and attempt to slice. PrusaSlicer attempts the slice, but after approximately 5 seconds, crashes to the desktop. It appears to occur regardless of other settings, such as ironing. Have attempted to replicate the issue in 2.2.0 and 2.3.0 Alpha 3, but those versions slice the objects without the error.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

fe60, 2020-11-25:
I can't reproduce the problem. Slicing works fine for me with all layer heights (alpha4 on Windows 10 build 19041.630).
Can you please add two files to the issue:
1. Prusa Slicer Project File (.3mf) with a configuration that makes the alpha4 failing on your machine
2. text file containing the system information (Help=>System Info, copy to clipboard, store in .txt and attach)

Gordontgopher, 2020-11-25:
See issue #5229, I suspect this is the same root cause, if so, it's been fixed and will be in the next release.

FidelCapo, 2020-11-27:
I think this is duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5208
This should be fixed in next release.
Closing.

closed 2020-11-27 by FidelCapo