prusa3d/PrusaSlicer #5258: Arrange does not take sequential printing into account anymore  [arrange] [regression]

mild-lemon, 2020-11-24:
### Version
2.3.0-alpha4

### Operating system type + version
macOS 10.15.7 "Catalina"

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro (stock firmare)

### Behavior
* Problem: Automatic placement function in 2.3.0-alpha4 does not take "sequential printing" distance into account anymore. This is a REGRESSION in comparison to 2.3.0-alpha3.

* Steps needed to reproduce the problem
  * enable "Sequential printing: Complete idividual objects" in "Print settings"
  * import object from .stl file
  * copy object
  * paste object
  * click "Arrange"

* Expected Results: As a result of clicking "Arrange", objects shall be placed in sufficent distance to comply with sequential printing extruder spacing settings (works in 2.3.0-alpha3!)
* Actual Results: As a result of clicking "Arrange", objects are placed in small distance only (presumable complying with normal printing spacing setting), PLUS the error message "Insufficiant distance" is displayed
[image]

_Is this a new feature request?_
This is NOT a new feature request.

#### Project File (.3MF) where problem occurs
[einkaufschip_1.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5593400/einkaufschip_1.3mf.zip)

tamasmeszaros, 2020-11-25:
Thanks for the report. The good news is that you can set the distance in the arrange menu by right clicking on the arrange button.

mild-lemon, 2020-11-28:
Bug still present in 2.3.0-beta1.

bubnikv, 2020-11-28:
BTW it is not regression. AFAIK It never did take sequential print into
account.

mild-lemon, 2020-11-28:
Both in 2.2.0 and up to 2.3.0-alpha3, "Arrange" would respect the value of extruder_clearance_radius if complete_objects was set to "true".
So yes, it is a REGRESSION.

closed 2020-12-01 by tamasmeszaros

bubnikv, 2020-12-05:
We have split the parameters for the arrangement spacing to 3 independent values:
FDM
FDM for sequential print
SLA

The default value of FDM for sequential print shall again respect the clearance radius.
The fix will be contained in PrusaSlicer 2.3.0-beta2.

commit 8027f660 (2020-12-01): Fix arrange distance with enabled sequential printing

fixes #5258