prusa3d/PrusaSlicer #5259: Monotonic infill leaves gaps in top infill where rectilinear does not

robnee, 2020-11-25:
### Version
Build:     PrusaSlicer-2.3.0-alpha4+win64-202011211615
Installed extensions:

### Operating system type + version

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 16,977MB
OpenGL installation
GL version:   4.6.0 NVIDIA 432.00
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 750 Ti/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S

### Behavior
Monotonic infill leaves gaps in top infill where standard Rectilinear infill does not
* Open attached 3MF file (two involute gears) and compare top infill on preview screen for gaps
* Right hand gear (rectilinear) has solid top infill
* Left hand gear (monotonic) has gaps

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
includes screenshot and 3MF models
[monotonic gaps.zip](https://github.com/prusa3d/PrusaSlicer/files/5593551/monotonic.gaps.zip)

bubnikv, 2020-11-25:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/5225

It does not look right indeed.

bubnikv, 2020-11-25:
Fixed with e5b3af0e533eaffa575820e57cfda10d597b9666
Thanks for the report.

closed 2020-11-25 by bubnikv

robnee, 2020-11-25:
You're very welcome.  Thank you.