prusa3d/PrusaSlicer #5260: Adaptive Cubic Infill issue

morimorida, 2020-11-25:
### Version
PrusaSlicer 2.3.0 Alpha 4

### Operating system type + version
Windows 10 64 bit

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S + MMU2S

### Behavior
When adaptive cubic infill is used, seldom the sloped infill fails and create a bogger on the surface.  See pictures below.

[image]
[image]
[image]

This was (actually still is printing) using PETG filament.  Is it possible to improve the logic or structure of it?  Maybe infill wall lines too thin?

bubnikv, 2020-11-25:
Is it sliced with alpha3 or alpha4? Alpha4 anchors the inner infill lines
to the longer ones, this issue should be mitigated by the anchors.
It may happen that the infill lines tear one by the other when crossing,
but I think yours is the issue of missing anchors.

morimorida, 2020-11-28:
I did slice with alpha4, I don't know what the length of the infill anchor should be so I just picked the default 1000 (unlimited).  Can you briefly explain this setting and what's the outcome of none vs unlimited?

bubnikv, 2020-11-28:
The adaptive cubic infill introduces lines, that start and end inside the infill. These lines were not anchored with short perpendicular line segments yet in alpha3. But on your picture it looks like the continuous infill lines fail at crossings.

sslupsky, 2021-01-12:
I encounter this (or similar) issue recently as well sliced with RC3.  The infill fails and creates a print distortion.
[image]

jalabb, 2021-03-23:
I don't know if it's related but Adaptive Cubic is generating infill in thin air for me : 
[image]
model is Hogwarts by hannah150303 on thingiverse, 5% infill

sslupsky, 2026-09-02:
Did not see anything in the release notes that addresses this.