prusa3d/PrusaSlicer #5261: Option to disable text replacement on MacOS

janis-ax, 2020-11-25:
### Version
2.2.0+

### Operating system type + version
Mac OS Big Sur - 11.0.1 (20B29)

### Behavior
I have text replacement on my MacBook enabled. So my MacBook replaces 'on' with 'online' and so on. 
The problem is, that the printer and filament settings have the option customized G-Code and in the text field Start-G-Code my MacBook replaces words like 'on'.  

In consequences, during slicing PrusaSlicer throws in error cause the G-Code in these customized sections is wrong. 

* _Steps needed to reproduce the problem_
&rarr;  enable text replacement on MacOS for the word 'la', 'on' or 'min' and the G-Code in theses section change.

* _Expected Results_
An option to disable text replacement for PrusaSlicer only or that there is no replacement

_Is this a new feature request?_
**Yes**

bubnikv, 2020-11-25:
I would bet we solved that with a wxWidgets patch, but it seems the current
wxWidgets we are using has changed.

https://github.com/wxWidgets/wxWidgets/commit/8d42890df49564d6d81e413bcd4428bf6a8b10e6

For reference (and for @yusanka), this is the methods provided by wxWidgets
3.1.3

```
void wxNSTextViewControl::CheckSpelling(bool check)
{
    if (m_textView)
        [m_textView setContinuousSpellCheckingEnabled: check];
}

void wxNSTextViewControl::EnableAutomaticQuoteSubstitution(bool enable)
{
    if (m_textView)
        [m_textView setAutomaticQuoteSubstitutionEnabled:enable];
}

void wxNSTextViewControl::EnableAutomaticDashSubstitution(bool enable)
{
    if (m_textView)
        [m_textView setAutomaticDashSubstitutionEnabled:enable];
}
```

bubnikv, 2020-12-07:
Update:

We are already disabling quote substitution and dash substitution. We need likely to call

- (void)setAutomaticSpellingCorrectionEnabled:(BOOL)flag
- (void)setAutomaticTextReplacementEnabled:(BOOL)flag

but for that we would have to patch our wxWidgets.

tjhorner, 2025-04-24:
Just ran into this on my Mac. I have text replacements for some characters that are used in the default g-code templates:

[image]

And they are replaced by simply visiting the "Custom G-code" tab.

closed 2026-09-29 by prusaslicer-bot[bot]