prusa3d/PrusaSlicer #5263: Segmentation fault on macOS when importing STL - PrusaSlicer 2.3.0-alpha4

oryanmoshe, 2020-11-25:
### Version
PrusaSlicer 2.3.0-alpha4

### Operating system type + version
macOS Catalina Version 10.15.6

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S
Firmware 3.9.1

### Behavior
I used the slicer yesterday, and it worked just fine.
Then, when trying to slice another model it crashed.
Now, whenever I open the slicer and try to import an STL file it crashes with `Segmentation fault: 11`, both when I try "drag & drop" and clicking the import button.
Here's a part of the error message:
```
Crashed Thread:        18  slic3r_tbb_7

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [37164]
```

oryanmoshe, 2020-11-25:
Just realized I had "Background processing" turned on.
When turning it off, I am able to load the STL file, but upon slicing it crashes again.

oryanmoshe, 2020-11-25:
**Update**
It seems to happen only when I try to slice this model, although it did work yesterday:
https://www.thingiverse.com/thing:4645349

oryanmoshe, 2020-11-25:
Also happens with another model **but only when I turn it upside down**

jaapp, 2020-11-25:
Seems to be similar to https://github.com/prusa3d/PrusaSlicer/issues/5229

mikelane, 2020-11-26:
For the latest alpha 4+, every model I've tried to slice so far has caused the app to crash.

alexbuicescu, 2020-11-27:
Changing the infill to `cubic` or `adaptive cubic` fixes the crash (solution taken from [this comment](https://github.com/prusa3d/PrusaSlicer/issues/5283#issuecomment-734466059)). Sometimes this does not work and I also have to change the layer height to something != 0.1 (0.15 and 0.2 work fine)

FidelCapo, 2020-11-27:
I think this is duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5208
This should be fixed in next release.
Closing.

closed 2020-11-27 by FidelCapo