prusa3d/PrusaSlicer #5264: Printer does not wait for nozzle/bed to cool down before it starts printing

fe60, 2020-11-25:
### Version
PrusaSlicer-2.2.0+win64-202003211145
PrusaSlicer-2.3.0-alpha4+win64-202011211615

### Problem
Usually the printer must wait until the nozzle and the bed have been heated up to the required temperature. This results in the following commands in the gcode:
```
M190 S60 ; wait for bed temp
M109 S215 ; wait for extruder temp
```

There is at least one scenario, where the printer has to wait until the nozzle/bed cooled down.
* last print was with PETG
* unload PETG (heats nozzle up to 230 degrees and bed to 85 degrees)
* load PLA
* start print with PLA
The nozzle cools down rather quickly and reaches the expected 215 degrees, when the print starts.
But the bed is still at almost 80 degrees, while its temperature should be 60 degrees.

### Proposal
Use the machine code to wait for cooling or heating instead of waiting for heating only.
`R` instead of `S`
https://marlinfw.org/docs/gcode/M109.html
https://marlinfw.org/docs/gcode/M190.html

_Is this a new feature request?_
No

foreachthing, 2020-11-25:
Just add it yourself. You can add that to your machine's custom gcode. But I guess you already knew that... 

Maybe you can even make it conditional to PLA?!

fe60, 2020-11-26:
I was not aware of the fact, that this part of the gcode is customizable.
@foreachthing Thanks for this hint.

closed 2020-11-26 by fe60