prusa3d/PrusaSlicer #5266: cursed wall thickness: 1.6mm on a circular object with 0.4mm nozzle? infill issue?

Mirarkitty, 2020-11-25:
### Version
at least 2.2.0 to 2.3.0 alpha 4

### Operating system type + version
Linux, appimage

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
* 0.4mm nozzle, default settings, 
* 1.6mm wall in a circle
* inner line (between the two exterior lines) gets erratic, lots of retractions, very slow, and seems to put in too much material -
* resulting print is uneven and stringy

#### Project File (.3MF) where problem occurs
[wheel_infill_foo.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/5597203/wheel_infill_foo.3mf.gz)

I'm guessing this is the result of a floating point problem, where something is going back and forth between one line and two lines.

2mm works perfectly fine (two interior lines all the way, all four lines filled fast and correct).

Mirarkitty, 2020-11-25:
[image]

Base is >4mm, it's quite easy to spot where the 1.6mm wall starts. Two left are 1.6mm (see .3mf above), the one to the right 2mm thickness.

bubnikv, 2020-11-25:
In your 3MF the only layer I see over extrusion is the following:
[image]

This over-extrusion disappears if you enable "Detect thin walls".

There is some over extrusion due to infill-perimeter overlap. Also you may want to calibrate your extruder, you may be over-extruding in general.

Mirarkitty, 2020-11-28:
Well, that would be a clue to the bug. I don't think it should need to be checked though?

Printer is calibrated.

bubnikv, 2020-11-28:
I asked @rtyr to print it.
The bug I have found only applies to a single layer, while you show an over-extrusion over the whole print.

rtyr, 2020-11-28:
Your 3MF project (slightly modified 0.3mm DRAFT, old LA1.0 K value):
[image]

Stock settings (0.3mm DRAFT/Prusament PLA/MK3S).
[image]

Printed on MK3S. FW3.9.1/PrusaSlicer Alpha4+62.

Mirarkitty, 2020-11-29:
Strange! I've also been able to print the same model earlier without issue, that's the weird bit here (and why I tried backing the slicer version). I'll generate some broken gcode, maybe that will be a clue.

Mirarkitty, 2020-11-30:
[broken_gcode_pla_extr_0.gcode.gz](https://github.com/prusa3d/PrusaSlicer/files/5615889/broken_gcode_pla_extr_0.gcode.gz)

This is the gcode for the PLA print above. (MMU2, printing PLA in first slot, might be a bit hot settings for a default printer.) (Since the problem was both with PETG and PLA I already ruled out filament settings.)

If that gcode works great on another printer, I'm even more confused.

rtyr, 2020-11-30:
[image]

> might be a bit hot settings for a default printer
> 
Your printer is not stock?

bubnikv, 2020-12-01:
Wrong firmware? Bondtech vs. the MK2 hobbled bolt extruder wheels? Extruder
steps off?

rtyr, 2020-12-06:
No response. G-code is ok. Closing.

closed 2020-12-06 by rtyr

Mirarkitty, 2020-12-08:
Factory assembled Prusa MK3S with switched nozzle (Nozzle X). Interesting. Very repeatable for me.

rtyr, 2020-12-08:
I saw weird unusual things which were caused by loose silicon sock on the heater.