prusa3d/PrusaSlicer #5267: Ironing is still buggy

rocketmany2k, 2020-11-25:
### Version
2.3.0 alpha4

_Use `About->About PrusaSlicer` for release versions_
Version 2.3.0-alpha4+win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 64-bit Professional

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S

### Behavior

Ironing does not work as expected. I have two issues:

1. What does "All solid surfaces" under Ironing Type mean? I would expect ironing on all visible surfaces. For example, if I choose this option for a cube, I expect all 6 surfaces get ironed. Is this right?  I printed a C-shaped object but only the top surface has been ironed. I can see infills and perimeters on the bottom surface. Is this a bug? How do I iron the bottom surface to make it looks (without holes and gaps among infills/perimeters) solid like the top?

2. Even the top surface has bug. I still see lots of holes and gaps between perimeters on the top surface. In addition, only 1/3 of the surface looked ironed smooth. The remaining 2/3 looks scratched and when I looked closely, I could see diagonal lines. I guess ironing in alpha4 does not work as well as ironing in SuperSlicer.

Please fix these issues

bubnikv, 2020-11-25:
Please provide 3MFs and screenshots.

rtyr, 2021-01-28:
> What does "All solid surfaces"

It should iron all solid surfaces. It is not implemented yet, it will be fixed in the next PS release. There is already related issue https://github.com/prusa3d/PrusaSlicer/issues/5703.

> Even the top surface has bug. I still see lots of holes and gaps between perimeters on the top surface.

You have likely heat creap issues (clogging). This can happen during the ironing, especially when using PLA filaments. It is mentioned in the PrusaSlicer documentation and also mentioned in different issues (see "ironing" labeled issues).

I am closing this issue, since there was no response.

closed 2021-01-28 by rtyr