prusa3d/PrusaSlicer #5269: External perimeter extrusion printed BEFORE others perimeters despite external_perimeters_first = 0

graizzaro, 2020-11-25:
### Version
2.3.0-alpha4+win32-202011211604
2.3.0-alpha3+win32-202011041521
2.3.0-alpha1+win32-202010171237
2.2.0+win32-202003211132

### Operating system type + version
Windows 7 32 bit (build 7601, Service Pack 1)

### 3D printer brand / version + firmware version (if known)
3DRag - Velleman K8200
firmware 3Drag Marlin V12_LCD_Alfa_2EXT175

### Behavior
In this print, external perimeter extrusion of internal triangle side, is printed BEFORE others perimeters despite ; external_perimeters_first = 0
while external perimeter extrusion of external triangle side, is made in right way after others perimeters.

You can see for example layer 28 but all layer are printed the same way.
[image]

#### Project File (.3MF) where problem occurs

[Triang 2020-1109 triangolo nuovo h21.zip](https://github.com/prusa3d/PrusaSlicer/files/5597988/Triang.2020-1109.triangolo.nuovo.h21.zip)

bubnikv, 2020-12-08:
I can reproduce it. Somehow PrusaSlicer does not like the constellation of holes. There are 4 holes in that single slice: one big in the center and three smaller for the screws.

If I close the smaller holes, no issue.

[image]

bubnikv, 2020-12-08:
When I patched just the top the hole with a cylinder, the order of perimeters is correct.

[image]

[Triang 2020-1109 triangolo nuovo h21 - patched.zip](https://github.com/prusa3d/PrusaSlicer/files/5659996/Triang.2020-1109.triangolo.nuovo.h21.-.patched.zip)

I have to admit that I thought I aware of most of the PrusaSlicer issues, but this one is new to me.

graizzaro, 2020-12-09:
Thanks for your interest.
It was just a case I discovered this problem.

I hope you can solve it

closed 2026-09-29 by prusaslicer-bot[bot]