prusa3d/PrusaSlicer #5272: in Prusaslicer 2.3.0-alpha4, when opening a .3MF project using "Open as project" the displayed Printer-profile name can mismatch the actual active Printer-profile

jaapp, 2020-11-25:
### Version
PrusaSlicer 2.3.0-alpha4 release

### Operating system type + version
MacOs Mojave 10.14.6

### Behavior

1. Start prusaslicer
2. Use configuration assistant to set up MK3S MMU2S profile:
[image]
And the following filament Profile Selection:
[image]
[image]

3. Update al profiles to latest version
4. Close prusaslicer
5. Copy custom Print and Printer profiles from 2.2.0 to 2.3.0-alpha4
The selected files are copied over
[image]
6. Start Prusaslicer
7. Acknowledge the message that the old Printer profile will be split into a Printer profile and an insantance
Result:
[image]
9. This custom printer has no filament profiles yet. So I copy a random filament, while removing the "Compatible print profiles condition" and clicking "Detach from system preset"
[image]
10.Final result:
Print profiles:
[image]

Filament profiles:
[image]
(!) Note how there are *no system presets*
11. Create and save a .3MF project with a "Print quality"- and Filament-profile selected which only should match with the PRUSA printer
12. Select a different printer profile
13. Drag previous created .3MF project onto prusaslicer and select the "Open as project" option
The Prusa buildplate is now shown with the Prusa dimensions. Also the Print- and Filament-profiles corresponding with the prusa mk3s are shown. But the "Printer * Schijtprinter" profile is selected
[image]
The MK3SMMU2S printer name is used in the generated g-code filename:
[image]
14. When selecting a different printer profile and then selecting "Printer 1 * schijtprinter" again the correct build plate is shown and the correct Print- and Filament-profiles are visible:
[image]

rtyr, 2020-11-25:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/5178.

bubnikv, 2020-12-04:
e42e25c implements:

Storing the physical_printer_settings_id into the 3MF, AMF, GCode.
Activating the physical_printer_settings_id when loading from 3MF, AMF, GCode.
The physical printer is only activated if it references the printer_settings_id
loaded from the same file.
When loading the presets from 3MF, AMF, GCode, the "external" profiles
are no more created for profiles which differ from the local profiles
the loaded profiles reference. Instead, the referenced profile is activated
and modified with the loaded preset. If the referenced profile does not
exist, but the profile refers to a system profile with the "inherits"
fileds, the system profile is loaded and modified instead.
This works for all profiles with the exception of multi-extruder
printer with multiple filament profiles modified. In that case
the first modified filament profile will be loaded as modified,
while the other modified profiles will be loaded as "external".
I believe this fixes your issue.

closed 2020-12-04 by bubnikv