prusa3d/PrusaSlicer #5273: compatibility pronlem wit Norton security

Mack1945, 2020-11-25:
### Version
_Version of PrusaSlicer used goes here_2.3.0-Alpha4

_Use `About->About PrusaSlicer` for release versions_Alpha4

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_ 

### Operating system type + version Windows 10
_What OS are you using, and state any version #s_ 
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_ 

### 3D printer brand / version + firmware version (if known) 
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_ 

### Behavior
* _Describe the problem_Norton Life Lock , after you run Alpha4 you can not turn on or off in Norton
* _Steps needed to reproduce the problem_insyall Alpha4 run it , you will see that the auto- protect is grayed out but its on with an * comment these settings cannot be changed due to a pending remediation action. 
  * _If this is a command-line slicing issue, include the options used_no
* _Expected Results_ auto-detect should be able to be turned on or off
* _Actual Results_ cant do anything to auto-protect

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_no screan shot

_Is this a new feature request?_ no 

#### Project File (.3MF) where problem occurs 
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_ 
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_ 
[image]

bubnikv, 2020-11-26:
Thanks for heads up.

What actions do you expect from us?
IMHO you should rather complain to your antivirus vendor.

Mack1945, 2020-11-26:
The info. that I sent you was just to inform you of a problem. What to do
with the information I sent you, is up to you all as to what to do with it
.Sorry that I bothered you with that information.

bubnikv, 2020-11-26:
> The info. that I sent you was just to inform you of a problem. What to do
with the information I sent you, is up to you all as to what to do with it
.Sorry that I bothered you with that information.

Thanks. I did not mean to be sarcastic. Still I would appreciate if you could please report this incident to your antivirus vendor.

bubnikv, 2020-12-07:
Norton security issue, it needs to be fixed at Norton side.
Closing.

closed 2020-12-07 by bubnikv