prusa3d/PrusaSlicer #5275: [Feature Request] Choose the amount of sides addable objects can have

bigbear2nd, 2020-11-25:
### Version
Version 2.3.0-alpha4+win64
### Operating system type + version
Windows64
### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S
#### 
_Is this a new feature request?_
Yes
####

When adding Modifiers or other Shapes to the Model, instead of giving "Cubus" or "Cylinder", it would be great to be able to choose the amount of sides the object should have. (additional to top and bottom)
The difference between Cubus and Cylinder is only the amount of sides.
With this option, the user could create Triangle, Square, .. Hexagon ... and so on based Objects. (3 sides to unlimited)

jaapp, 2020-11-26:
I assume you mean surfaces?

Defining a shape by the number of surfaces alone is not enough information.
1 surface: sphere, donut, etc...
2 sufrace: cone, hemisphere, etc...
3 surface: cylinder, tetrahedron, etc...
4 surface: cube, square pyramid, etc...

neophyl, 2020-11-26:
If you want to add different shapes as modifiers then just use a modelling program to generate the ones you want and then use them instead of the built in ones.  There is the option to load external models for the modifiers.

foreachthing, 2020-11-26:
You could use Blender. It's fast, simple and powerful if you need more complex shapes.

With the "3D Print Toolbox" addon you can export those meshes with one click.
Here are some standard meshes you can add very quickly:
[image]

neophyl, 2020-11-26:
Funnily enough Blender with the basic mesh objects was precisely what I was thinking of.

foreachthing, 2020-11-26:
> Funnily enough Blender with the basic mesh objects was precisely what I was thinking of.

I love using the monkey as modifier!! 😆😉

rtyr, 2021-02-25:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/4926.

bubnikv, 2021-07-26:
Modifier gallery implemented in the upcoming PrusaSlicer 2.4.0-alpha1.

closed 2021-07-26 by bubnikv