prusa3d/PrusaSlicer #5279: errors related to libslic3r_gui

ogiwolf, 2020-11-26:
### Version
master 
### Operating system type
Windows 10
### Behavior

I follow the step from "Noob guide (step by step)"
if works totally fine until i press the Start Without debugging button
it have 1357 errors, and it all related to the file "libslic3r_gui"

I have no idea how to fix this problem. 
Is that i miss installing something?

foreachthing, 2020-11-26:
See here for hints:
* https://github.com/prusa3d/PrusaSlicer/issues/5096
* https://github.com/prusa3d/PrusaSlicer/issues/5039

ogiwolf, 2020-11-26:
> See here for hints:
> 
> * #5096
> * #5039

Thanks for replying
I have tried the batch filed u supply in #5039 
while i am running the first batch file, it run out lots of error related to the libslic3r_gui

such as :

 "C:\Users\James\Desktop\PSlicer\PrusaSlicer\build\PrusaSlicer.sln" (rebuild 目標) (1) ->
       "C:\Users\James\Desktop\PSlicer\PrusaSlicer\build\ALL_BUILD.vcxproj.metaproj" (Rebuild 目標) (4) ->
       "C:\Users\James\Desktop\PSlicer\PrusaSlicer\build\src\slic3r\libslic3r_gui.vcxproj.metaproj" (Rebuild 目標) (25) -
       >
       "C:\Users\James\Desktop\PSlicer\PrusaSlicer\build\src\slic3r\libslic3r_gui.vcxproj" (Rebuild 目標) (71) ->
       (ClCompile 目標) ->
         C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(90,22): error C2001: 常數中包含新行字元 (正在編譯原始程式
       檔 C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\AboutDialog.cpp) [C:\Users\James\Desktop\PSlicer\Pru
       saSlicer\build\src\slic3r\libslic3r_gui.vcxproj]
         C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(98,7): error C2062: 未預期的類型 'Slic3r::GUI:
       :GUI_App' (正在編譯原始程式檔 C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\AboutDialog.cpp) [C:\Users\James\
       Desktop\PSlicer\PrusaSlicer\build\src\slic3r\libslic3r_gui.vcxproj]
         C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(90,22): error C2001: 常數中包含新行字元 (正在編譯原始程式
       檔 C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\ConfigSnapshotDialog.cpp) [C:\Users\James\Desktop\PS
       licer\PrusaSlicer\build\src\slic3r\libslic3r_gui.vcxproj]
         C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(99,1): error C2143: 語法錯誤: 遺漏 ';' (在 '{'
       之前) (正在編譯原始程式檔 C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\AboutDialog.cpp) [C:\Users\James\Deskto
       p\PSlicer\PrusaSlicer\build\src\slic3r\libslic3r_gui.vcxproj]
         C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(90,22): error C2001: 常數中包含新行字元 (正在編譯原始程式
       檔 C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\KBShortcutsDialog.cpp) [C:\Users\James\Desktop\PSlic
       er\PrusaSlicer\build\src\slic3r\libslic3r_gui.vcxproj]
         C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(98,7): error C2062: 未預期的類型 'Slic3r::GUI:
       :GUI_App' (正在編譯原始程式檔 C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\ConfigSnapshotDialog.cpp) [C:\Use
       rs\James\Desktop\PSlicer\PrusaSlicer\build\src\slic3r\libslic3r_gui.vcxproj]
         C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(99,1): error C2143: 語法錯誤: 遺漏 ')' (在 ';'
       之前) (正在編譯原始程式檔 C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\AboutDialog.cpp) [C:\Users\James\Deskto
       p\PSlicer\PrusaSlicer\build\src\slic3r\libslic3r_gui.vcxproj]
         C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(98,7): error C2062: 未預期的類型 'Slic3r::GUI:
       :GUI_App' (正在編譯原始程式檔 C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\KBShortcutsDialog.cpp) [C:\Users\
       James\Desktop\PSlicer\PrusaSlicer\build\src\slic3r\libslic3r_gui.vcxproj]
         C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(99,1): error C2143: 語法錯誤: 遺漏 ';' (在 '{'
       之前) (正在編譯原始程式檔 C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\ConfigSnapshotDialog.cpp) [C:\Users\Jam
       es\Desktop\PSlicer\PrusaSlicer\build\src\slic3r\libslic3r_gui.vcxproj]
         C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(99,1): error C2447: '{': 遺漏函式標頭 (舊樣式型式清單
       ?) (正在編譯原始程式檔 C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\AboutDialog.cpp) [C:\Users\James\Desktop
       \PSlicer\PrusaSlicer\build\src\slic3r\libslic3r_gui.vcxproj]
         C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(99,1): error C2143: 語法錯誤: 遺漏 ';' (在 '{'
       之前) (正在編譯原始程式檔 C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\KBShortcutsDialog.cpp) [C:\Users\James\
       Desktop\PSlicer\PrusaSlicer\build\src\slic3r\libslic3r_gui.vcxproj]
         C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(90,22): error C2001: 常數中包含新行字元 (正在編譯原始程式
       檔 C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\3DScene.cpp) [C:\Users\James\Desktop\PSlicer\PrusaSl
       icer\build\src\slic3r\libslic3r_gui.vcxproj]
         C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(99,1): error C2143: 語法錯誤: 遺漏 ')' (在 ';'
       之前) (正在編譯原始程式檔 C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\ConfigSnapshotDialog.cpp) [C:\Users\Jam
       es\Desktop\PSlicer\PrusaSlicer\build\src\slic3r\libslic3r_gui.vcxproj]
         C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(304,1): error C2143: 語法錯誤: 遺漏 ';' (在 '&'
        之前) (正在編譯原始程式檔 C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\AboutDialog.cpp) [C:\Users\James\Deskt
       op\PSlicer\PrusaSlicer\build\src\slic3r\libslic3r_gui.vcxproj]
         C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(90,22): error C2001: 常數中包含新行字元 (正在編譯原始程式
       檔 C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\GLShadersManager.cpp) [C:\Users\James\Desktop\PSlice
       r\PrusaSlicer\build\src\slic3r\libslic3r_gui.vcxproj]
         C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(99,1): error C2143: 語法錯誤: 遺漏 ')' (在 ';'
[... cut ...]
       mes\Desktop\PSlicer\PrusaSlicer\build\src\slic3r\libslic3r_gui.vcxproj]
         C:\Users\James\Desktop\PSlicer\PrusaSlicer\src\slic3r\GUI\AboutDialog.cpp(40,16): error C2664: 'Slic3r::GUI::D
       PIAware<wxDialog>::DPIAware(wxWindow *,wxWindowID,const wxString &,const wxPoint &,const wxSize &,long,const wxS
       tring &)': 無法將引數 1 從 'wxStandardID' 轉換為 'wxWindow *' [C:\Users\James\Desktop\PSlicer\PrusaSlicer\build\src\slic3
       r\libslic3r_gui.vcxproj]

Thanks again for helping me

foreachthing, 2020-11-26:
Your Windows is in english? 
Why I ask? See this reply from @sorrowfeng:
> I reinstalled an English version of windows and it solved the problem, is it the coding?
> Thank you very much for your help.

ogiwolf, 2020-11-26:
> Your Windows is in english?
> Why I ask? See this reply from @sorrowfeng:
> 
> > I reinstalled an English version of windows and it solved the problem, is it the coding?
> > Thank you very much for your help.

My windows is in Chinese...
So i have to reinstall windows to english?
Is there another way to solve it?

Thanks for the quick reply

sorrowfeng, 2020-11-26:
> > 您的Windows是英文的吗？
> > 我为什么要问？看到来自@sorrowfeng的回复：
> > > 我重新安装了英文版的Windows，它解决了这个问题，是编码吗？
> > > 非常感谢您的帮助。
> 
> 我的窗户是中文的...
> 所以我必须将窗户重新安装成英文吗？
> 还有另一种解决方法吗？
> 
> 感谢您的快速答复

重装中文版的Windows都没用, 装英文版的Windows 再 加上稳定的科学上网网速, 运行bat文件就可以了

ogiwolf, 2020-11-26:
> > > 您的Windows是英文的吗？
> > > 我为什么要问？看到来自@sorrowfeng的回复：
> > > > 我重新安装了英文版的Windows，它解决了这个问题，是编码吗？
> > > > 非常感谢您的帮助。
> > 
> > 
> > 我的窗户是中文的...
> > 所以我必须将窗户重新安装成英文吗？
> > 还有另一种解决方法吗？
> > 感谢您的快速答复
> 
> 重装中文版的Windows都没用, 装英文版的Windows 再 加上稳定的科学上网网速, 运行bat文件就可以了

看到中文格外親切XD
想問一下那我之後安裝英文版的windows 可以包成執行檔給其他非中文的windows上面運行嗎 

不好意思，我是新手 所以問題很多QQ

sorrowfeng, 2020-11-26:
> > > > 您的Windows是英文的吗？
> > > > 我为什么要问？看到来自@sorrowfeng的回复：
> > > > > 我重新安装了英文版的Windows，它解决了这个问题，是编码吗？
> > > > > 非常感谢您的帮助。
> > > 
> > > 
> > > 我的窗户是中文的...
> > > 所以我必须将窗户重新安装成英文吗？
> > > 还有另一种解决方法吗？
> > > 感谢您的快速答复
> > 
> > 
> > 重装中文版的Windows都没用, 装英文版的Windows 再 加上稳定的科学上网网速, 运行bat文件就可以了
> 
> 看到中文格外親切XD
> 想問一下那我之後安裝英文版的windows 可以包成執行檔給其他非中文的windows上面運行嗎
> 
> 不好意思，我是新手 所以問題很多QQ

可以的, 只是编译环境需要, 我觉得是编译出文件的编码问题

ogiwolf, 2020-11-26:
> 可以的, 只是编译环境需要, 我觉得是编译出文件的编码问题

好的那我去測試看看，非常感謝你們拯救我這個小菜鳥>"<

foreachthing, 2020-11-26:
Use virtulabox and install a new windows on it - that's what I do.

ogiwolf, 2020-11-26:
> Use virtualbox and install a new windows on it - that's what I do.
Thanks for the advice!

After i run through the two batch file you provide.
I open the PrusaSlicer.sln in visual studio 2019.
and then i click the start without debugging.

It shows :

System OpenGL library released
Loading MESA OpenGL library: C:\Users\ogiwo\Desktop\PSlicr\PrusaSlicer\build\src\Debug\mesa\opengl32.dll
MESA OpenGL library was not loaded
PrusaSlicer.dll was not loaded

C:\Users\ogiwo\Desktop\PSlicr\PrusaSlicer\build\src\Debug\prusa-slicer-console.exe (process 7996) exited with code -1.
Press any key to close this window . . .

foreachthing, 2020-11-26:
Interesting!
I don't have a `mesa` folder in my debug?!
[image]

_Maybe_ because I have CUDA SDK installed? I don't even have `opengl32.dll` in my `PrusaSlicer` folder anywhere. I don't think I can help you there - sorry!

ogiwolf, 2020-11-27:
Thanks anyway 👍 
You already help a lot!

ogiwolf, 2020-11-27:
> Interesting!
> I don't have a `mesa` folder in my debug?!
> [image]
> 
> _Maybe_ because I have CUDA SDK installed? I don't even have `opengl32.dll` in my `PrusaSlicer` folder anywhere. I don't think I can help you there - sorry!

Sorry to bother you.
I reinstall it, and this time it didn't show PrusaSlicer.dll was not loaded
But still show PrusaSlicer.dll was not loaded

Do u know how to fix this problem?

foreachthing, 2020-11-27:
I actually never use VS to rebuild.
I do it via cmd line `msbuild -ds -nologo PrusaSlicer.sln -t:rebuild -m -p:Configuration=Release -verbosity:normal`.

VS is rebuilding (debug) right now, though....

Ok! Building with VS2019 Pro:
<pre>Loading MESA OpenGL library: C:\dev\PrusaSlicer\build\src\Debug\mesa\opengl32.dll
MESA OpenGL library was not loaded
PrusaSlicer.dll was not loaded

C:\dev\PrusaSlicer\build\src\Debug\prusa-slicer-console.exe (process 9320) exited with code -1.
To automatically close the console when debugging stops, enable Tools->Options->Debugging->Automatically close the console when debugging stops.
Press any key to close this window . . .</pre>

What's going on with VS???

foreachthing, 2020-11-27:
`Release` works - sort of.
I get it to compile but it still complains about opengl32. Can't run it, because the VM has no real graphics card. Even with `--sw_renderer` it does not start. But, it works just fine on the host computer.

ogiwolf, 2020-11-27:
> I actually never use VS directly to rebuild.
> I do it via cmd line `msbuild -ds -nologo PrusaSlicer.sln -t:rebuild -m -p:Configuration=Release -verbosity:normal`.
> 
> VS is rebuilding (debug) right now, though....
> 
> Ok! Building with VS2019 Pro:
> 
> Loading MESA OpenGL library: C:\dev\PrusaSlicer\build\src\Debug\mesa\opengl32.dll
> MESA OpenGL library was not loaded
> PrusaSlicer.dll was not loaded
> 
> C:\dev\PrusaSlicer\build\src\Debug\prusa-slicer-console.exe (process 9320) exited with code -1.
> To automatically close the console when debugging stops, enable Tools->Options->Debugging->Automatically close the console when debugging stops.
> Press any key to close this window . . .
> What's going on with VS???
`

> `Release` works - sort of.
> I get it to compile but it still complains about opengl32. Can't run it, because the VM has no real graphics card. Even with `--sw_renderer` it does not start. But, it works just fine on the host computer.

It finally works!
The opengl32 issue is due to the virtual machine.
And the PrusaSlicer.dll I klick the "rebuild solution" and it work!
Thank u very much for spending time to help me.
I am very appreciated.

closed 2020-11-27 by ogiwolf