prusa3d/PrusaSlicer #5282: Alpha 4 keeps crashing - Here is a 3MF file causes a crash

jweaver100, 2020-11-26:
### Version
2.3.0-alpha4+Win64

### Operating system type + version
Windows10

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior
Like most people, the 1st time I used Alpha4 it crashed and it pretty much does this constantly, but I havnt tracked down the cause, but I suspect there are lots of triggers.

Today I tried to slice a model that I have printed many times and it crashes... Its fine in Alpha3, so I figued it was worth creating this post as I have a reproducable example of how to crash Alpha4

#### Project File (.3MF) where problem occurs
Attached
[crashes alpha 4.zip](https://github.com/prusa3d/PrusaSlicer/files/5601812/crashes.alpha.4.zip)

2Average, 2020-11-26:
This file also crash Alpha4, try to slice it, a few % in, it collapses.
https://cdn.thingiverse.com/assets/e8/6b/fb/d2/17/12_BallSocket_Joint.stl

FidelCapo, 2020-11-26:
Thank you for report. This crash should be fixed in next release.
Closing.

closed 2020-11-26 by FidelCapo