prusa3d/PrusaSlicer #5283: Prusa Slicer alpha4 crashs on Mac OS 11.0.1

ClausHat, 2020-11-26:
### Version
_2.3.0 alpha 4_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

_mac OS Big Sur 11.0.1_

 used the slicer yesterday, and it worked just fine.
Then, when trying to slice another model it crashes.
Now, whenever I open the slicer and try to import an STL file it crashes.
slicing works until I change the "physical printer" profile.
after changing the physical printer profile and I press the slice button -->  slicer crashes.

Following message from Mac OS crash report:

Process:               PrusaSlicer [13431]
Path:                  /Applications/PrusaSlicer 2.3.0 alpha 4.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.3.0-alpha4+-202011211604 (PrusaSlicer-2.3.0-alpha4+-202011211604)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           PrusaSlicer [13431]
User ID:               501

Date/Time:             2020-11-26 15:23:48.342 +0100
OS Version:            macOS 11.0.1 (20B50)
Report Version:        12
Anonymous UUID:        B446AC53-F93A-7D42-41E1-88262F28CEE1

Sleep/Wake UUID:       E729ACB5-CDCC-4E6C-A1BF-B8F876FC3776

Time Awake Since Boot: 140000 seconds
Time Since Wake:       920 seconds

System Integrity Protection: enabled

Crashed Thread:        11  slic3r_tbb_1

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [13431]

VM Regions Near 0x10:
--> 
    __TEXT                      1002f9000-101d74000    [ 26.5M] r-x/r-x SM=COW  /Applications/PrusaSlicer 2.3.0 alpha 4.app/Contents/MacOS/PrusaSlicer

Thread 0:: slic3r_main  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff20307e7e mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff203081f0 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff20434be7 __CFRunLoopServiceMachPort + 316
3   com.apple.CoreFoundation      	0x00007fff204332ba __CFRunLoopRun + 1315
4   com.apple.CoreFoundation      	0x00007fff204326be CFRunLoopRunSpecific + 563
5   com.apple.HIToolbox           	0x00007fff2869efd0 RunCurrentEventLoopInMode + 292
6   com.apple.HIToolbox           	0x00007fff2869edcc ReceiveNextEventCommon + 709
7   com.apple.HIToolbox           	0x00007fff2869eaef _BlockUntilNextEventMatchingListInModeWithFilter + 64
8   com.apple.AppKit              	0x00007fff22c4bf85 _DPSNextEvent + 883
9   com.apple.AppKit              	0x00007fff22c4a74b -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1366
10  com.apple.AppKit              	0x00007fff22c3cbda -[NSApplication run] + 586
11  com.prusa3d.slic3r/           	0x00000001017facae wxGUIEventLoop::OSXDoRun() + 174
12  com.prusa3d.slic3r/           	0x000000010195a6e7 wxCFEventLoop::DoRun() + 39
13  com.prusa3d.slic3r/           	0x0000000101892d99 wxEventLoopBase::Run() + 137
14  com.prusa3d.slic3r/           	0x000000010185a66d wxAppConsoleBase::MainLoop() + 189
15  com.prusa3d.slic3r/           	0x000000010185a2a9 wxAppConsoleBase::OnRun() + 25
16  com.prusa3d.slic3r/           	0x000000010156c6fb wxAppBase::OnRun() + 59
[... 552 lines cut ...]
mapped file                      245.0M       34 
shared memory                       44K        5 
===========                     =======  ======= 
TOTAL                              2.0G     1636 
TOTAL, minus reserved VM space     1.6G     1636 

Model: Macmini7,1, BootROM 426.0.0.0.0, 2 processors, Dual-Core Intel Core i5, 2,8 GHz, 8 GB, SMC 2.24f32
Graphics: kHW_IntelIrisItem, Intel Iris, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x483943434E4E4E424C54424C41522D4E5444
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x483943434E4E4E424C54424C41522D4E5444
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x13B), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1669)
Bluetooth: Version 8.0.1f5, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Serial ATA Device: APPLE SSD SM0128G, 121,33 GB
Serial ATA Device: APPLE HDD HTS541010A9E662, 1 TB
USB Device: USB 3.0 Bus
USB Device: Keyboard Hub
USB Device: Mass Storage Device
USB Device: Apple Keyboard
USB Device: IR Receiver
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 2.0 Hub
USB Device: MX530 series
Thunderbolt Bus: Mac mini, Apple Inc., 26.1

BastR, 2020-11-26:
I confirm the crash under macOs Big Sur. For me it happens when I switch from any infill  to a gyroid OR if I slice a model with gyroid infill already selected

ClausHat, 2020-11-27:
> I confirm the crash under macOs Big Sur. For me it happens when I switch from any infill to a gyroid OR if I slice a model with gyroid infill already selected

Thank you very much Alex, I changed the infill from gyroid to another and now it works like before.

FidelCapo, 2020-11-27:
I think this is duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5208
This should be fixed in next release.
Closing.

closed 2020-11-27 by FidelCapo