prusa3d/PrusaSlicer #5285: Slice Issue

moreirae, 2020-11-26:
###  
PrusaSlicer-2.3.0-alpha4

### 
Windows 10 64bits
### Creality Ender 3 Pro / Marlin 2.0.7

### 
When I try print this kind of Vase with 0% Infil and 2 shells the slicer create some solid parts, see the picture, how i can avoid this?
[image]

_

moreirae, 2020-11-26:
the .3mf file
[Twist v6.zip](https://github.com/prusa3d/PrusaSlicer/files/5604787/Twist.v6.zip)

rtyr, 2020-11-27:
I can see it is not sliced in vase mode, which would be probably better in this case.

Steps to fix your issue:
Your model is not solid, so you should start by repairing it (right-click on the model).
[image]

Then disable "Ensure wall thickness option".
[image]

It should fix your issue. You may try regular `vase mode` with thicker extrusion width instead for better surface finish.

[image]

Can we close this issue?

neophyl, 2020-11-27:
iirc the OP is printing a 'vase' with 2 shells.  Not printing in 'vase mode'.  The issue is the undesired extra solid infill on the sloped areas which has also been reported before and already has an issue number.  Unless my memory is playing tricks on me.

I only post as the original poster opened a thread on the forum where I pointed out this was a known issue.  I couldn't find the already open issue at the time though on the forum.  I think https://github.com/prusa3d/PrusaSlicer/issues/1054 is the one I was remembering.

rtyr, 2020-11-27:
Yes. Steps I mentioned above will help in this particular case though.

[image]

moreirae, 2020-11-27:
thanks too all, can close the issue

closed 2020-11-27 by rtyr