prusa3d/PrusaSlicer #5286: 2.3.0-alpha4+win64 - fill bed with instances issues

shwekwek, 2020-11-26:
### Version
2.3.0-alpha4+win64

### Operating system type + version
Win64
[pixel_5.1.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5604786/pixel_5.1.3mf.zip)

### 3D printer brand / version + firmware version (if known)
MK3s

### Behavior
**Fill bed with instances:** seems like it totally ignore the tool path. it's filling the bed completely till the edges, and when you slice - it shows the tool path error.

**Arrange options > Gap size:** there is a bar to adjust the gap size, but you can type in parameter manually. it jumps between random(?) parameters, and I couldn't get the needed on, which is (for example) 1mm.

project file attached, as well as prtscn of the problems.

[image]

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

tamasmeszaros, 2020-12-08:
I think this relates to #3477

lukasmatena, 2023-04-02:
Arrange function accounts for brim and skirt since [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). This also solves the issue with fill bed with instances. Closing.

closed 2023-04-02 by lukasmatena