prusa3d/PrusaSlicer #5290: Alpha4 crash with 0.6 nozzle printer variant only

davidhbrown, 2020-11-27:
### Version
Version 2.3.0-alpha4+win64

### Operating system type + version
Windows 10 Version 20H2

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK2.5S 0.6 nozzle

### Behavior
* Brief freeze and crash to desktop when green progress bar shows about 1/3 through slicing
* _Steps needed to reproduce the problem_
  * Set Printer to Original Prusa i3 MK2.5S 0.6 nozzle 
  * Import `Kitchenaid_Knob.stl` from https://www.thingiverse.com/thing:257150
* _Expected Results_: mesh is sliced
* _Actual Results_: Brief freeze and crash to desktop

This file does not report any defects and slices fine in alpha4 if I use the regular or 0.25 nozzle variations of the printer. It also slices fine with the 0.6 nozzle in 2.2.0+win64

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[Kitchenaid_Knob.zip](https://github.com/prusa3d/PrusaSlicer/files/5606101/Kitchenaid_Knob.zip)

fe60, 2020-11-27:
Was it really about the size of the nozzle?
59628d0 includes a fix related to the infill pattern https://github.com/prusa3d/PrusaSlicer/issues/5229#issuecomment-732370015

lukasmatena, 2020-11-27:
I did not reproduce it in current master. I DID reproduce in alpha4, it disappears when infill is disabled. That makes it likely a duplicate of #5208 (and #5229), which should be solved.

Closing.

closed 2020-11-27 by lukasmatena

davidhbrown, 2020-11-27:
> 
> 
> Was it really about the size of the nozzle?
> [59628d0](https://github.com/prusa3d/PrusaSlicer/commit/59628d0221c49a6801bdbb3b3e9f6b0c589220a3) includes a fix related to the infill pattern [#5229 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/5229#issuecomment-732370015)

Very possibly not; the size of the nozzle is just what different in the visible name of the failing printer configuration vs. other printer configurations provided by Prusa for this machine. The "Printer" choices appropriate to my machine are:
* Original Prusa i3 MK2.5S
* Original Prusa i3 MK2.5S 0.25 nozzle
* Original Prusa i3 MK2.5S 0.6 nozzle

Only the third of these (I currently have an 0.6 nozzle mounted) failed with this mesh, though it was fine with several other meshes.