prusa3d/PrusaSlicer #5293: Screen capture from within slicer?

reddawgesq, 2020-11-27:
PrusaSlicer-2.3.0-alpha4+win64
Windows 10
Prusa i3 MK3S/Mini, current F/W versions
New feature request

Would like to be able to take screeshots from within the slicer program without having to resort to third-party capture programs. Suggest an option under File/Export/Export Plate As with further options of JPG or PNG image format whereupon the ability to draw a wire frame box around the desired portion of the screen would present.

bubnikv, 2020-11-27:
Sorry, but screenshot functionality is implemented directly into Windows 10. It is overkill to implement such functionality into slicer.

closed 2020-11-27 by bubnikv

reddawgesq, 2020-11-27:
True, but the PrtScr function captures the entire screen or window that frequently contains unwanted/private information such as the user's taskbar and filepath information that then has to be subsequently removed using Gimp or a similar image manipulation program. If Win10 has a built-in wire-frame-like area selection capability for capture purposes, I am unaware of it. Please correct me if such exists. Thank you.

Sent from my Verizon, Samsung Galaxy smartphone

foreachthing, 2020-11-27:
[windows] + [shift] + [s] will do that.

reddawgesq, 2020-11-27:
My my my, what WILL they think of next!? TYVM.