prusa3d/PrusaSlicer #5294: Seam painting configuration exported to 3mf is opaque, undocumented and poorly-designed

tg73, 2020-11-27:
### Version
2.3.0-alpha3

### Operating system type + version
Win10 x64

### Behavior

I need to generate seam painting configuration programmatically alongside my programmatically-generated models. My program emits a 3mf file tailored to suit PS. This is an example of what PS itself exports inside `3dmodel.model` when seam painting is present:

```xml
<triangle v1="965" v2="966" v3="967" slic3rpe:custom_seam="000000004006003003400401600034054424443444041A44334440412400A4054463000340544A4044041234404464304045002000401633300300300303000000000401A003044A40544A04002300450020003303000401240544A0450063444344002000463000304040544244041A300004012044A045006300400603330004006444041240544A30030304404160034440020004634400401643300004240544A400634444041A4434344041A04500A440544230044041603440544644300401244433330000044244041204500A40544630003040500A400A04054041230033030004404160030030034004644300004240544A400630044041603300030004240544A4006340450020004016300440416034004016443300034400401643444054464304400204500A04463344041A400A44054423004404160340544600442400A300424443304444041A4304500200030004016000333330030033" />
```

This meaning of the data in `@slic3rpe:custom_seam` is entirely opaque, and is not in keeping with the intent of the 3mf format and general principles of good interoperability when using XML. This data should be broken down into smaller pieces of meaningfully marked-up XML so that its meaning is clear, and so that it can be read and generated by external programs. This extension of the 3mf format should also be documented to enhance interoperability.

bubnikv, 2020-11-28:
> Seam painting configuration in 3mf format is opaque, undocumented and poorly-designed

Poorly designed? Why such strong words? You are judging design choices that you don't understand.

> This meaning of the data in @slic3rpe:custom_seam is entirely opaque, and is not in keeping with the intent of the 3mf format and general principles of good interoperability when using XML. This data should be broken down into smaller pieces of meaningfully structured and marked-up XML so that its meaning is clear, and so that it can be read and generated by external programs.

The purpose of the slic3rpe:custom_supports string is to describe recursive splitting of large triangles and to define which sub-triangles are painted with which color (for which purpose). The decision we made was driven by the requirement to 

1) not modify the input model
2) to limit the size increase of the 3MF file and to limit writing / parsing times.

An XML format is not very efficient for storing large amount of data.

If you paint the whole model with support enforcers, you will find that all the triangles will be annotated with 
slic3rpe:custom_supports="4"

for the blockers it is 
slic3rpe:custom_supports="8"

for the custom seam it is again
slic3rpe:custom_seam="4"

> This extension of the 3mf format should also be documented to enhance interoperability.

The day has 24 hours, we have 1738 open issues, each of them very important to somebody. There is so many things we "should be" doing.

tg73, 2020-11-28:
@bubnikv I apologise for and regret my use of unnecessarily strong words, I understand your frustration, and I am grateful for your courteous response.  Let explain myself better:

My expectations of Prusa's posture regarding 3MF is informed by [this blog post](https://blog.prusaprinters.org/3mf-file-format-and-why-its-great_30986/), which encourages data sharing and interoperability between applications. This expectation drove my disappointment at the opaque and non-self-descriptive representation.

XML is indeed rather "bulky". But the core [3MF specification](https://3mf.io/specification/) accepts this, and mitigates some aspects of the inefficiency by using zip compression. It would be easy for the core 3MF spec to encode all the vertex and triangle data far more efficiently, such as using base64 encoding of array memory - but 3MF wants the data to be accessible. One of the core stated aims of the 3MF Consortium is that the data should be human readable, which opaque custom encodings are not. Avoiding custom encodings also allows generic toolchains such as XSLT to work with the data. A more structured XML representation also serves to describe the data, reducing or removing the need for explanatory documentation.

I have spent many years working professionally as a developer with creating and manipulating XML-based representation of structured information - often with XML files several hundred megabytes in size. I appreciate your concerns regarding file size and writing/parsing times, although I would argue that zip compression will mitigate much of the file size increase, and modern XML parsers are remarkably efficient. I would be surprised if the use of a more structured representation would impact the processing time of 3MF read/write operations to such a showstopping extent that a custom encoding is the only choice, also given that 3MF read/write operations are surely infrequent for most common usage patterns of PS as a whole.

Thank you for the partial description of the slic3rpe:custom_seam attribute. I can try to study the source code to understand the encoding of the recursive splitting, but if you are able to give me even a very terse explanation assuming an expert audience that might well be enough.

tg73, 2020-11-29:
I've tested emitting `slic3rpe:custom_seam="4"`. The correct areas appear shaded blue in seam painting mode, but the enforcement area does not seem to be respected when slicing (with 2.3.0 beta1). I can't tell if this is a 3MF parsing issue, or a more general seam painting/slicing issue. Attached are: `a0.3mf` - this is one of my programmatically-generated 3MF's, and `project.3mf` which is the saved PS project after importing `a0.3mf`. Screen shots below. I've moved the object from its imported 0,0 origin because the axes indicator thing in the UI blocks the view of the object and I don't see how to turn it off.

[image]

[image]

[files.zip](https://github.com/prusa3d/PrusaSlicer/files/5612326/files.zip)

lukasmatena, 2020-11-29:
> if you are able to give me even a very terse explanation assuming an expert audience that might well be enough.

@bubnikv was faster, but I can at least give some details about the encoded information. The characters are limited to hexadecimal digits (0-9A-F), each digit is one triangle, the string shall be read backwards (i.e., last digit in the string is the parent triangle). To decode the digit, convert the value to binary (you'll get four binary digits) and split them in two pairs. This gives you two decimal numbers XY (range [0-3]). The meaning is:

    Y = how many edges shall be split in half
    X = (Y==0): NOTHING (0) / ENFORCER (1) / BLOCKER (2)
        (Y==1): index of the edge which should be split (0/1/2)
        (Y==2): index of the edge which should NOT be split (0/1/2)
        (Y==3): 0 (unused)

For example, 048A means: 0xA=0b10'10=(2,2) => two edges of the triangle shall be split, edge opposite vertex number 2 will be left alone. The triangle therefore has three children, their data follow. They are 0x8=0b10'00, 0x4=0b01'00, 0x0=00'00 => neither is further split, one is an enforcer, one is a blocker, one is nothing special. The compression could have been better, but I wanted to keep the digit-triangle correspondence and stick to hex digits and not any other weird characters. Some combinations are not used (at least not now).

See TriangleSelector.cpp and functions `perform_split`, `serialize` and `deserialize` for more details.

> correct areas appear shaded blue in seam painting mode, but the enforcement area does not seem to be respected when slicing

Thanks, I'll investigate that.

tg73, 2020-11-29:
@lukasmatena Thanks very much for the explanation. I think your encoding is really quite elegant. The resulting division of a triangle with one edge split and all edges split seems obvious. However, what exactly happens when two edges are split? In meshes which are not limited to triangles, this would typically produce a triangle and a quad. But I assume we're sticking to triangles here, so the resulting division is not so clear to me.

lukasmatena, 2020-11-29:
@tg73 
> However, what exactly happens when two edges are split?

One of the following. The `perform_split` function should tell which one (I believe it depends on the order of the vertices).
[image]

bubnikv, 2020-11-30:
The painting tool splits the triangles in a way to produce triangles closer
to equilateral with each splitting iteration. The splitting stops when a
limit on maximum edge length is reached. The whole gymnastics is there to
achieve triangle painting and splitting at an interactive speed while
keeping the number of newly created triangles low and to avoid
re-triangulation of split triangles when unmarking regions of a triangle
and then marking different regions of the same triangle.

We may consider to apply distance field tricks in the future to smoothen
the painted edges even more.
https://steamcdn-a.akamaihd.net/apps/valve/2007/SIGGRAPH2007_AlphaTestedMagnification.pdf

bubnikv, 2020-11-30:
BTW what are you working on?

tg73, 2020-11-30:
Thanks both for the very helpful explanations. The distance field paper is fascinating - many years ago I worked on a video editing product, including developing some pixel shaders for video effects, so reading the paper was a pleasant flashback!

@bubnikv These are parts for a variation on a kind of Swedish musical instrument called a [nyckelharpa](https://en.wikipedia.org/wiki/Nyckelharpa), which I play. The instrument has a keyboard with a key for each semitone. My variation, which I'm inventing/developing/prototyping, allows any pitch to be played (like a violin) by using hundreds or thousands of thin blades sandwiched together that project the profile of the fingertip onto the string. There is a strong microtonal tradition in Swedish folk music for violin, which I'd like to be able to explore as a nyckelharpa player. If you look at a picture of a regular nyckelharpa, you will see that the keys vary in length along the scale. Also the clearance away from the string increases along the scale to account for the deflection of the string when vibrating. Likewise, every single blade has to be slightly different. I generate the blades from C# code using the various 3D geometry libraries exposed by [MatterControl](https://github.com/MatterHackers/MatterControl). The little tip insert, which is what contacts the string, is printed in TPU. The body of each blade will be ASA or perhaps PC-Blend, which I'll try when the satin steel sheets are available - then I could even have the TPU tip extended to the first layer. The tiny text which uniquely labels each blade is rendered as a single extrusion width mesh by rendering an old single-line pen plotter font. My C# code renders blades, arranges them in plates (accounting for wipe tower), emits a 3MF for each plate, then runs PS console to slice, then post-processes the gcode, notably to add my own ironing moves over the top of each blade - I did that before PS had ironing, but I still don't want any extrusion while ironing, which PS can't do (also ironing over multiple top surface materials in one pass). This is ironing for engineering reasons, not cosmetic - I need the top surface to be smooth, without any tiny zits. I'm using The Zero Beast's slightly modified extruder because the stock MK3 extruder is too hard (impossible?) to calibrate the IR sensor chimney for printing hard and flex materials in the same MMU2S print. Maybe this will be better with the MK3S+ extruder.

tg73, 2020-12-03:
@lukasmatena I just wanted to check if the problem below is still on your radar. I am happy to log a specific issue if that is helpful wrt issue tracking.
>>correct areas appear shaded blue in seam painting mode, but the enforcement area does not seem to be respected when slicing

>Thanks, I'll investigate that.

bubnikv, 2020-12-05:
We hope that @lukasmatena fixed your issues where the seam was not always respected.

closed 2022-01-11 by lukasmatena