prusa3d/PrusaSlicer #5295: Slicer crashes during slicing of specific stl-files

FalcoGoodbody, 2020-11-27:
### Version
V2.3.0-alpha4+win64

### Operating system type + version
Win10 20H2 (19042.630)

### 3D printer brand / version + firmware version (if known)
Creality Ender3 Silent Board 1.1.8

### Behavior
when slicing a specific stl file (https://cults3d.com/en/3d-model/art/3d-filament-wizard-for-shane-s-birthday) the slicer crashes without error message.

[3d-filament-wizard-for-shane-s-birthday20201020-8172-gkd8l5.zip](https://github.com/prusa3d/PrusaSlicer/files/5608911/3d-filament-wizard-for-shane-s-birthday20201020-8172-gkd8l5.zip)

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs

1WileCoyote, 2020-11-27:
Version
V2.3.0-alpha4+win64

Operating system type + version
Win10 20H2 (19042.572)

3D printer brand / version + firmware version (if known)
Prusa I3 MK3S + MMU2S

Behavior
slicing starts, progress bar goes up to approx. 25% then application ends without message.
In the left lower corner "preparing infill..." was shown.
If I set infill to 0% slicing completes, seems to be an infill related issue.
Slicing also succeeds, if I change the infill pattern from gyroid to rectilinear (honeycomb works as well).
[Cleo the Citten.zip](https://github.com/prusa3d/PrusaSlicer/files/5609214/Cleo.the.Citten.zip)
.

dgkimpton, 2020-11-27:
I get the same issue, I've simplified my reproduction case a lot to a single 3mm thick wall on a cylinder. 

Set print settings to: **0.2mm SPEED @MK3**
Set filament to: **Prusament PLA**
Set printer settings to: **Original Prusa i3 MK3S**

Change **Layers and Perimeters | Perimeters** to **3**
Change **Infill | Fill density** to **10%**
Change **Infill | Fill pattern** to **Gyroid**

Add STL **[prusa-test.zip](https://github.com/prusa3d/PrusaSlicer/files/5609709/prusa-test.zip)**
Click **Slice now**

** * poof * ** crash happens.

Windows event viewer adds the following
`Faulting application name: prusa-slicer.exe, version: 2.3.0.0, time stamp: 0x5fb93e3d
Faulting module name: PrusaSlicer.dll, version: 0.0.0.0, time stamp: 0x5fb93e37
Exception code: 0xc0000005
Fault offset: 0x00000000009621cf
Faulting process ID: 0x127c
Faulting application start time: 0x01d6c50a67051d7e
Faulting application path: C:\PrusaSlicer-2.3.0-alpha4+win64-202011211615\prusa-slicer.exe
Faulting module path: C:\PrusaSlicer-2.3.0-alpha4+win64-202011211615\PrusaSlicer.dll
Report ID: a865cb96-4d0c-4004-b5af-94c1d0e4dff5`
[windows error report.zip](https://github.com/prusa3d/PrusaSlicer/files/5609744/windows.error.report.zip)

Slicer version: 2.3.0-alpha4-win64
Windows version: 10 (2004 build 19041.630)
Printer: Prusa i3 Mk3S

Janek250, 2020-11-28:
me too, with several stl files

obelisk, 2020-11-28:
I could not reproduce this on 59628d0221c49a6801bdbb3b3e9f6b0c589220a3 on macOS so it might be a windows specific issue.

1WileCoyote, 2020-11-28:
fixed in beta1, please close.

closed 2020-11-28 by bubnikv