prusa3d/PrusaSlicer #5300: Filename incorrect when exporting G-Code with MMU. Always refers filament type in extruder 1

GinoBMW, 2020-11-28:
### Version
_Version of PrusaSlicer

2.3.0-Alpha4

### Operating system type + version

Windows 10 64bit

### 3D printer brand / version + firmware version (if known) 

MK3S - MMU2
_
### Behavior

When Exporting G-Code the filament type in the filename does not represent the type of filament being used being used. it always refers to Extruder 1 no matter what extruder you have chosen to print with.

E.g. If extruder 1 is set for PETG and you are using extruder 5 that is PLA the filename will show PETG in its description even though you are not using PETG.   

_Is this a new feature request?_

No, Bug fix

bubnikv, 2020-11-28:
```
output_filename_format =
{input_filename_base}_{layer_height}mm_{filament_type[0]}_{printer_model}_{print_time}.gcode
output_filename_format =
{input_filename_base}_{nozzle_diameter[0]}n_{layer_height}mm_{filament_type[0]}_{printer_model}_{print_time}.gcode
```

The filament type reference
```filament_type[0]```
references the 1st extruder independent from whether it prints or not.
I tried a quick prototype of adding a new "initial_tool" placeholder to the
output file name field, but that failed to produce a valid template before
the slicing finished. Also the initial tool may not be a good placeholder
here, because it may print soluble support, which may not what you are
looking for. Likely we will add a new placeholder like
"first_object_filament_type", but it will not be a 5 minutes job due to
sequential / non-sequential printing etc.

novski, 2021-07-22:
I have the same problem. 
`{input_filename_base}_{layer_height}mm_{filament_type[4]}_{printer_model}_{print_time}.gcode`
Writes the 5 Extruder (spool) Material to the Filename. 
`_{filament_type}_` -> error
`_{filament_type[]}_` -> error
`_{filament_type[*]}_` -> error
`_{filament_type[x]}_`-> error
`_{filament_type[i]}_`-> error
`_{filament_type[n]}_`-> error
I can't figure out the wildcard or how to get the acutely chosen Material in to the filename. 
If anyone has a solution to this, please let me know.

bubnikv, 2022-01-10:
Implemented with a591d9e9fe6dd7a1a5605688c3ce07f1225c4c5e

Implemented two new placeholders: "initial_filament_type" and "printing_filament_types".

closed 2022-01-10 by bubnikv