prusa3d/PrusaSlicer #5301: [Bug] 2.3.0-alpha4+linux-x64 opens new instance when pressing ctrl-i

RichiH, 2020-11-28:
### Version
2.3.0-alpha4+linux-x64

### Operating system type + version
Debian testing / unstable

### 3D printer brand / version + firmware version (if known)
Prusa Mini+ 4.2.1-final+2072

### Behavior
Pressing ctrl-i in Prusa Slicer is supposed to open the import dialogue. Instead, it starts a new instance of Prusa Slicer.

RichiH, 2020-11-29:
Also present in beta1

tehcaster, 2020-12-03:
Same here, on openSUSE Tumbleweed with KDE. I've tried the official releases with and without GTK3 in name. Also own build from git main branch.

Here's console output from the GTK build:

[after prusa-slicer is executed, many similar messages ending with the following]
...
(prusa-slicer:8524): GLib-GObject-CRITICAL **: 17:00:00.089: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:8524): GLib-GObject-WARNING **: 17:00:00.089: invalid (NULL) pointer instance

(prusa-slicer:8524): GLib-GObject-CRITICAL **: 17:00:00.089: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:8524): GLib-GObject-WARNING **: 17:00:00.089: invalid (NULL) pointer instance

(prusa-slicer:8524): GLib-GObject-CRITICAL **: 17:00:00.089: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:8524): GLib-GObject-WARNING **: 17:00:00.089: invalid (NULL) pointer instance

(prusa-slicer:8524): GLib-GObject-CRITICAL **: 17:00:00.089: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

[here ctrl+i is pressed]

(prusa-slicer:8655): GLib-GObject-WARNING **: 17:00:06.530: invalid (NULL) pointer instance

(prusa-slicer:8655): GLib-GObject-CRITICAL **: 17:00:06.530: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:8655): GLib-GObject-WARNING **: 17:00:06.530: invalid (NULL) pointer instance

(prusa-slicer:8655): GLib-GObject-CRITICAL **: 17:00:06.530: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:8655): GLib-GObject-WARNING **: 17:00:06.530: invalid (NULL) pointer instance

(prusa-slicer:8655): GLib-GObject-CRITICAL **: 17:00:06.530: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:8655): GLib-GObject-WARNING **: 17:00:06.530: invalid (NULL) pointer instance

(prusa-slicer:8655): GLib-GObject-CRITICAL **: 17:00:06.530: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
[2020-12-03 17:00:06.630332] [0x00007f4a2effd640] [error]   Not primary owner of DBus name - probably another PrusaSlicer instance is running.
[2020-12-03 17:00:06.630361] [0x00007f4a2effd640] [error]   Dbus Messages listening terminating.

(prusa-slicer:8655): GLib-GObject-WARNING **: 17:00:06.709: invalid (NULL) pointer instance

(prusa-slicer:8655): GLib-GObject-CRITICAL **: 17:00:06.709: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:8655): GLib-GObject-WARNING **: 17:00:06.709: invalid (NULL) pointer instance

(prusa-slicer:8655): GLib-GObject-CRITICAL **: 17:00:06.709: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

[similar output continues]

Same kind of stuff is printed (including the dbus message) when pruse-slicer is running and another instance is executed from different terminal. So it seems to be a consequence of ctrl+i starting another instance, not the cause.

tehcaster, 2020-12-05:
Oh, I see. There's a new menu item "Window/Open new instance CTRL+I" so the same shortcut is assigned to two actions. On Windows, the import "wins", while on Linux, it's the new instance.

nordurljosahvida, 2020-12-06:
@RichiH can you confirm your issue is already represented in this thread? https://github.com/prusa3d/PrusaSlicer/issues/4711

bubnikv, 2020-12-06:
We will do something in the next version, we did not decide yet what exactly we will do.

bubnikv, 2020-12-07:
Fixed with https://github.com/prusa3d/PrusaSlicer/pull/4734

closed 2020-12-07 by bubnikv