prusa3d/PrusaSlicer #5303: Crash on slicing

gallouglas, 2020-11-28:
### Version
2.3.0-alpha4-win64

### Operating system type + version
Windows 10, build 1809 LTSC

### 3D printer brand / version + firmware version (if known)
Modified Zonestar Z8S (300x300x300mm), firmware - Marlin 1.1.8

### Behavior
When you try to slice, the process crashes with the program closing itself.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

[5.zip](https://github.com/prusa3d/PrusaSlicer/files/5610385/5.zip)

1WileCoyote, 2020-11-28:
did you use gyroid infill?
If the slicing completes with rectilinear infill instead, this issue was probably solved in beta1 already.

rtyr, 2020-11-28:
Fixed in 2.3.0-beta1. Closing.

closed 2020-11-28 by rtyr