prusa3d/PrusaSlicer #5307: [Feature request / UX Discussion] Support enforcers should work in all modes

erispe, 2020-11-28:
### Version
_PrusaSclier-2.3.0-alpha3_

### Operating system type + version
_Windows 10_

### Behavior
So this is a request to rethink how Support Enforcers work in regards to the different support modes. The single sentence summary is that a support enforcer should always enforce supports, even if support mode is set to None, because that is what it's name implies. (If the user has enforcers in their scene and wants no supports to show up then they should be disabled by _hiding_ them and setting support mode to None)

I feel like the options "Supports on build plate only" and "Everywhere" or "None" should be the only three options and then you add your support enforcers and blockers on top of that. The "For support enforcers only" becomes redundant. I want I want to be able to use the "Supports on build plate only" together with custom support enforcers, which currently is impossible, and I need to set up a complicated scene in "Supports everywhere" mode where I block everything that's not directly above the build plate.

In summary I think it would be more powerful and cover more use cases if enforcers always work regardless of mode.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2021-08-23:
A notification will be shown in PrusaSlicer 2.4.0 (an alpha will be released shortly) to inform about having enforcers with supports turned off.

bubnikv, 2021-09-22:
no response, closing.-

closed 2021-09-22 by bubnikv