prusa3d/PrusaSlicer #5308: [Bug] 2.3.0-beta1 macOS: file menu, keyboard commands do not launch file dialogs

alecperkins, 2020-11-28:
### Version
PrusaSlicer-2.3.0-beta1+-202011281022

### Operating system type + version
macOS Catalina Version 10.15.6 (Build 19G2021)

### 3D printer brand / version + firmware version (if known)
n/a

### Behavior
- pressing `command ⌘` + `i` or clicking File > Import > Import STL/OBJ/AMF/3MF
Expected: import dialog is opened
Actual: nothing happens

- pressing `command ⌘` + `o` or clicking File > Open
Expected: open dialog is opened
Actual: nothing happens

- pressing `command ⌘` + `s` or clicking File > Save Project
Expected: save dialog is opened
Actual: nothing happens

The import menu and keyboard shortcut do work as expected in `PrusaSlicer-2.3.0-alpha4+-202011211604` on same OS. 
Fortunately, dragging and dropping files into the window still works.

#### Project File (.3MF) where problem occurs
n/a

obelisk, 2020-11-29:
I couldn't reproduce on master (also 30d7ef2), though I'm on Big Sur and maybe that's changing something.

bubnikv, 2020-11-30:
@alecperkins 
> Version
> PrusaSlicer-2.3.0-beta1+-202011281022
> Operating system type + version
> macOS Catalina Version 10.15.6 (Build 19G2021)

Anything special about your setup? Hackintosh? Some unusual OSX configuration? Any unusual language or keyboard layout selected?

FidelCapo, 2020-11-30:
I tested it on Catalina 10.15.7 and this shortcuts works ok.

alecperkins, 2020-12-01:
Nothing unusual setup, a standard 16-in MBP, default US English keyboard. Only thing noteworthy is it's in clamshell mode with an external monitor. It still didn't work after a couple restarts. However, I then updated to macOS 10.15.7 (Build 19H15) and now it's working fine. Both the release and the version built from source. I'll close this since the fix seems like just being fully updated. Thanks anyway!

closed 2020-12-01 by alecperkins