prusa3d/PrusaSlicer #5309: New Project option unavailable if all objects have been deleted

erispe, 2020-11-28:
### Version
PrusaSlicer-2.3.0-alpha3

### Operating system type + version
Windows 10

### Behavior
* You are unable to create a 'New Project' from the 'File' menu if all the models have been deleted from the project. This can be an issue if you intended to make a new project but removed models manually first, as you then need to add an arbitrary model so that the option is available again.

* _Steps needed to reproduce the problem_
  * Open a .3mf project file and delete the models so the build plate is empty, then go to the "File" menu and observe that the

neophyl, 2020-11-28:
I may be missing something but aren't the results of deleting all the objects and clicking new project the same ?

You end up with an empty plater with the same presets selected.  Why would you want to click a control that does nothing ?

erispe, 2020-11-29:
Because with a new project you can hit ctrl+s to and you are prompted to save a new .3mf file. Otherwise if working in an old project you have to be careful to to chose "Save Project As..". It's a small thing, but I don't see why the program _shouldn't_ be consistent and always allow a ctrl+n.

neophyl, 2020-11-29:
Good point :)

Kachidoki2807, 2020-12-24:
Hello,

From my point of view, "New Project" should behave differently from "Delete All".
- Delete All: Only remove all the objects from the plater, keeping the settings as-is.
- New Project: Restart from a fresh state, like a new startup of PS, new 3mf, no object on the plater nor modified settings.

I am often annoyed when I want to reset all the setup to default when switching from a project to a completely different one (a really new project), to have to manually discard the print/filament/printer profiles. Since the 2.3 and its fast startup, it is even quicker to close and repoen PS than creating a "new project"...

nordurljosahvida, 2021-01-03:
Exactly, completely agree with @neophyl and @Kachidoki2807, new project should be a blank slate, purge all settings, cut off legacy from last saved file to avoid accidentally saving into it and therefore overwriting it. It should literally refresh the session like a clean start up.

bubnikv, 2021-01-04:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/5671
We will get there one day. The current behavior is a result of multiple refactorings of the old Slic3r.

lukasmatena, 2021-09-02:
I think that this is solved in 2.4.0-alpha1. Delete All does the same as if one removes all the objects one by one, the action is taken as a modification of currently opened project. New Project is available even immediately after the delete.
Closing.

closed 2021-09-02 by lukasmatena