prusa3d/PrusaSlicer #5310: Fill bed with instances does not accomodate for skirt

Nikoju, 2020-11-28:
### Version
PrusaSlicer 2.3.0 Beta 1

### Operating system type + version
Win 10 x64

### Behavior
I've noticed that when using the "fill bed with instances" option it fills the bed just fine but does not leave distance to the build plate's edges for the skirt.

_Is this a new feature request?_ NO

#### Project File (.3MF) where problem occurs

[Nameplate_Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/5610978/Nameplate_Issue.zip)

[image]

JohnOCFII, 2020-11-29:
Similar issue with raft/supports.
[image]

lukasmatena, 2023-04-02:
Arrange function accounts for brim and skirt since [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena